raspberrypi/rpi-imager #818: [FEATURE] Upgrade to Qt6.6 for macOS builds

tdewey-rpi, 2024-02-12:
**Is your feature request related to a problem? Please describe.**
Qt 6.4 isn't actively supported and is not an LTS release. Qt 6.6 is the only version that has active and current OSS support.

**Describe the solution you'd like**
Mandate Qt 6.6 to build the macOS rpi-imager. This will include raising the macOS deployment target to 10.16 / 11 / Big Sur.

**Describe alternatives you've considered**
Considered:
- No action: **Unacceptable** - CVEs are actively reported against Qt, and fixing on an unsupported version may expose users to these problems. On the upside, people running Macs over 12 years old could still use the latest release.
- Upgrade to Qt 6.5 LTS: **Unacceptable**. While getting commercial support, it doesn't get active OSS support, and rpi-imager is an OSS project. Potential CVE exposure would continue here.

**Additional context**
This was flagged by a user trying to build the source tree to investigate a bug - where they noticed that they couldn't build with Homebrew because the 6.6 version supplied by Homebrew makes use of the C++ filesystem API, which is not offered in Mac SDKs before 10.15 - and our deployment target was set to macOS 10.14.

Building for macOS11 may introduce other behaviours - eg - Dark Mode support in macOS. These should be mitigated as part of any PR.

lurch, 2024-02-12:
Will this change also bump the minimum Ubuntu version supported?

maxnet, 2024-02-12:
I think it is a bit strange to cite CVEs as an argument, but only upgrade Mac and not on Windows.

>On the upside, people running Macs over 12 years old could still use the latest release.

The thing about Macs is that the entry level models come with so little amount of disk space, that people that are stuck with those often cannot upgrade in practice.

tdewey-rpi, 2024-02-12:
> I think it is a bit strange to cite CVEs as an argument, but only upgrade Mac and not on Windows.

CVEs are an argument for the selection of the version - Qt 5.15 is still supported on Windows and Linux, while on macOS we were required to move to Qt6 for Apple Silicon.

> 
> > On the upside, people running Macs over 12 years old could still use the latest release.
> 
> The thing about Macs is that the entry level models come with so little amount of disk space, that people that are stuck with those often cannot upgrade in practice.

There's nothing stopping those users running 1.8.5 - and since 1.8.5 uses generated os_lists I'm much more comfortable with people remaining there if they run an unsupported OS.

maxnet, 2024-02-12:
>There's nothing stopping those users running 1.8.5

Not sure how easy that is to find for the average user.
Chances are they only visit the main raspberrypi.com download page, and conclude things do not work...

tdewey-rpi, 2024-02-12:
That's a risk we're going to have to take. My assessment of the telemetry is that this will practically impact an _extremely_ small number of users.

tdewey-rpi, 2024-02-27:
Edit title - this should really be Qt6.6 for everyone, where the operating floor will raise as follows:

macOS: From 10.15 to 11 (released 2020)
Linux: No effective change, still Ubuntu 22.04 (released 2022)
Windows: From Windows 7 to Windows 10 (released 2015)

tdewey-rpi, 2024-03-05:
Issue with the OS list identified as being dependent on the dimensions of the initial load of the OSPopup. Using the default window size causes no list to be drawn at all, with additional rendering bugs, but when fullscreen it draws OK every time after the first load, irrespective of size. Not entirely clear what's going wrong - there's nothing obvious in the debug log, so more digging required.

tdewey-rpi, 2024-09-11:
Qt6 now deployed on all 64-bit platforms.

closed 2024-09-11 by tdewey-rpi