raspberrypi/rpi-imager #821: [BUG]Imager hashes WiFi password, won't connect until manually changed back

DavidCC00, 2024-02-16:
**Describe the bug**
After imaging, the Pi fails to connect to wifi.  Opening the firstrun.sh and changing wifi password to un-hashed instance will allow connection 

**To Reproduce**
Raspberry Pi Imager v1.8.5 - Windows 10
Raspberry Pi 4 - OS Lite 64 Bit
Advanced Settings -> Configure Wireless LAN SSID:  XXXXXXX  Password:  Password  Hidden SSID  Wireless LAN Country: US
*close Raspberry Pi Imager*
Reopen RPI, go back through steps, look at Advanced Settings, check "show password" Password value:   d9830385f98e41f35841dcea471219c2913590b354a722bcaf347433fac1c396  (Hash of "Password") 

**Expected behaviour**
Password values not saved at all in RPI after being closed.

DavidCC00, 2024-02-16:
Nevermind, I thought you had to close it.  It just did it again on my newest image (I never closed it). 

[image]

lurch, 2024-02-17:
This has been asked about multiple times before ( https://github.com/raspberrypi/rpi-imager/issues?q=is%3Aissue+hashed ), and this is just how it's designed to work.

> After imaging, the Pi fails to connect to wifi. Opening the firstrun.sh and changing wifi password to un-hashed instance will allow connection

That suggests that maybe you made a typo in the details you entered into the Imager GUI?

Note that there _was_ a slight usability bug to do with entering passwords #531 which was fixed in RPi Imager v1.8.4
And there was also a bug with the WiFi configuration in Raspberry Pi OS Lite images #749 which was fixed in the December 11th 2023 Lite images.

tdewey-rpi, 2024-02-19:
I',m inclined to mark this as 'Won't fix', based on expected behaviour.

1. Saving WiFi passwords remains a popular feature, and short of a significant security issue I'm not inclined to remove that function
2. Hashing passwords has been discussed numerous times already, and I have yet to receive a compelling argument to change that behaviour.

Regarding your WiFi association failure, as @lurch points out, based on your description it appears that you must have made a typo entering the password - which happens to all of us at some stage or another. Did you also re-attempt to enter the password through Imager?

DavidCC00, 2024-02-19:
Something screwy is going on.  I'm wondering if it's a corner case bug from some unique combo of my pc hardware, it being windows 10, etc.  Not only am I having issues with the wifi, but after I ssh in, it always rejects the password, too.  I am not exaggerating when I say I tried this 30 times.  I have noticed that periodically I get different hashes despite always entering in the same password.  

>based on your description it appears that you must have made a typo entering the password

I can reveal the password for wifi in RPI, so that'seasy to eliminate.  Ive also used a keylogger to verify.  I also had a buddy who's real good (he's Software Engineer 3 for Amazon/ AWS) remote in, and he ran a script to hash the password and got a different result than the hash produced by RPI when running this linux command. 

````
echo 'ThePassword' | openssl passwd -6 -stdin
````

Shouldn't the hash stay the same, or does the Key change periodically?

Anyways, I'm buying a cable and will be doing this manually.  At this point, after 30 attempts, and no other reports, gunna assume I hit a bizarre corner case.

lurch, 2024-02-19:
> Shouldn't the hash stay the same, or does the Key change periodically?

It's a salted hash, to prevent replay attacks. https://en.wikipedia.org/wiki/Salt_(cryptography)

DavidCC00, 2024-02-20:
Discovered the problem.  The Pi was shipped to me defective, found traces bridged by a solder blob.

closed 2024-02-20 by DavidCC00