raspberrypi/rpi-imager #822: Imager on Win 11 is reporting expired certificate for downloads.raspberrypi.com

mj1z, 2024-02-17:
Downloading images via Imager on a win 11 pc from the raspberrypi.com website breaks with this error : 

[image]

This problem was first reported her as a Mac only problem, but unfortunately closed:
https://github.com/raspberrypi/rpi-imager/issues/661

The problem obviously both occurs on Mac's as well as on Pc's

The problem was also reported on raspberry.com:
https://forums.raspberrypi.com/viewtopic.php?t=357672

maxnet, 2024-02-17:
It does not claim anything has expired, just that your computer is unable to contact the server used to check if a certificate has been revoked.

Can you reach http://r3.o.lencr.org/ ?

mj1z, 2024-02-18:
@naxbet yes it connects, just a blank white page

mj1z, 2024-02-18:
> It does not claim anything has expired, just that your computer is unable to contact the server used to check if a certificate is in good standing or has been revoked.
> 
> Can you reach http://r3.o.lencr.org/ ?

yes it connects, just a blank white page

maxnet, 2024-02-18:
You mentioned downloads.raspberrypi.com in the title, and for that you would need to be able to reach the revokation server r3.o.lencr.org

However the IP in your screenshot does not belong to downloads.raspberrypi.com at all, but to github.
Github's SSL provider is digicert.
So it is probably http://ocsp.digicert.com or http://crl4.digicert.com instead.

mj1z, 2024-02-18:
> You mentioned downloads.raspberrypi.com in the title, and for that you would need to be able to reach the revokation server r3.o.lencr.org
> 
> However the IP in your screenshot does not belong to downloads.raspberrypi.com at all, but to github. So I am guessing you are trying to download something different than RPI OS. Github's SSL provider is digicert. So it is probably http://ocsp.digicert.com or http://crl4.digicert.com instead that it is unable to reach. Any kind of firewall software installed?

Yes I'm trying to download the Homebridge image for raspberry pi. I'm currently on my Mac so I can't check if the pc will connect to any of those url's until tomorrow (I will update you). I'm not aware of any firewall installed on that pc, it's a pc from my work place and it's locked down by some policies  but I don't think so. The Imager works fine when downloading the image manually and selecting it afterwords from the Imager and it will write to the card, only choosing the image directly from the Imager generets the error.

tdewey-rpi, 2024-02-19:
Unfortunately this may prove to be academic - the Homebridge OS has been removed from the OS image list after failing to respond to a request for information. Do you see the issue with any other OS download, @mj1z ?

mj1z, 2024-02-19:
> Unfortunately this may prove to be academic - the Homebridge OS has been removed from the OS image list after failing to respond to a request for information. Do you see the issue with any other OS download, @mj1z ?

No I haven't used Imager for any other purposes than for Homebridge. If what youre saying is true, then at least the Homebridge image should have been removed form the Imagers select list, it's very frustrating for the end user to go through all of this, and yes I know theres an alternative manuel way to download the image but that doesn't justify leaving broken stuff hanging like that. So please don't close the ticket before the Image is back on the repo or the image has been removed from the Imager

tdewey-rpi, 2024-02-19:
> No I haven't used Imager for any other purposes than for Homebridge. If what youre saying is true, then at least the Homebridge image should have been removed form the Imagers select list, it's very [...]

The reason for my comment is that Homebridge was removed this morning GMT.

mj1z, 2024-02-19:
> You mentioned downloads.raspberrypi.com in the title, and for that you would need to be able to reach the revokation server r3.o.lencr.org
>
> However the IP in your screenshot does not belong to [...]

Both urls connect from my work pc

mj1z, 2024-02-19:
> Both urls connect from my work pc

> The reason for my comment is that Homebridge was removed this morning GMT.

Ah okay I misunderstood your answer then, I hope the problem will be resolved soon.

tdewey-rpi, 2024-02-19:
At this point there's nothing to suggest there's an issue with `rpi-imager`. The reported error is one of being unable to verify the certificate of a third party hosted image - which unfortunately isn't something we can directly address.

Further muddying the waters is that I can't see this issue reported elsewhere - a failure of the SSL plugin would be causing all kinds of pandemonium, but I see no sign of that on the forums nor on this issue tracker. As extra salt, the OP's report is against a now-removed OS, though other OSes do host via the same underlying substrate.

As a result, I don't see what action we can take - the failure appears to be external, I cannot reproduce it, and I'm not seeing the expected large-scale reporting that would follow github's certificate chain being unverifiable. Every available point of evidence is therefore pointing to an issue with your network configuration or provider - and we're not in a position to debug the network of users of `rpi-imager`.

Finally, I'm not inclined to leave this issue open to be commandeered by another reporter with a different issue (as we so frequently see), which leaves me with one option: marking as 'Can't repro', and closing the issue.

Should we see more reports that are explicitly:
- Windows 11
- Latest `rpi-imager`
- Failure to check revocation for `github`

Then I will reopen the issue.

closed 2024-02-19 by tdewey-rpi