raspberrypi/rpi-imager #824: Question: RPI-Imager Wifi/Hostname configuration

hkfuertes, 2024-02-19:
Hello, I'm right now building a buildroot image for an specific project, and I would like to make my image compatible with configurator on the imager, mainly setting the hostname and the wifi info. How does the app do that? What would I need to do on my buildroot?

Sorry, I did not know where to ask this. And thank you!

tdewey-rpi, 2024-02-19:
Sounds like you are looking for this: https://cloud-init.io

Then take a look at how Canonical offer Ubuntu Server to see how they flag their cloud-init support.

Closing, as this isn't strictly an issue with the Raspberry Pi Imager codebase, and we don't yet have a mechanism for tracking documentation (and a blank GH issue isn't the route to do this).

closed 2024-02-19 by tdewey-rpi

lurch, 2024-02-20:
> Then take a look at how Canonical offer Ubuntu Server to see how they flag their cloud-init support.

...by which I think Tom means "take a look at the Ubuntu Server entries in https://downloads.raspberrypi.com/os_list_imagingutility_v4.json " :wink: