raspberrypi/rpi-imager #825: [BUG]: Weird partitioning of the disk

vsvjoirwv, 2024-02-22:
### What happened?

Hi, I am using Raspberry Pi 5 with the official RPi OS Bookworm x64. The system and the bootloader were updated to the latest versions. I bought an NVME Argon NEO 5 case and connected Verbatim Vi3000 2TB SSD to it. The SSD is recognised by the system:
`$ sudo lspci
0000:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0000:01:00.0 Non-Volatile memory controller: MAXIO Technology (Hangzhou) Ltd. NVMe SSD Controller MAP1202 (rev 01)
0001:00:00.0 PCI bridge: Broadcom Inc. and subsidiaries Device 2712 (rev 21)
0001:01:00.0 Ethernet controller: Device 1de4:000`

When I want to install the RPi OS on this SSD drive. The whole installation process goes absolutely normal. I expect to have two partitions on the SSD afterwards: one (512 MB) for boot and one (all available space minus 512 MB) for root. However, rpi imager creates one boot partition of the right size and one ext4 partition of size 4.93GB, with 100% of the space being used. The rest of the SSD stays unallocated.

I understand that there could be theoretical incompatibility with the PCIe HAT, but I am able to format the complete drive and create partitions on it via gparted, so it seems that the drive is recognized correctly. Below are screenshots of the drive after I used rpi imager. The whole situation seems pretty weird to me.
[image]
[image]
[image]
[image]

### Version

1.8.5 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

RPi OS Bookworm x64

### Selected OS

RPi OS Bookworm x64

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

NVMe SSD via Third-party PCIe attachment

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
$ rpi-imager
Custom config.txt entries: ""
Custom cmdline.txt entries: " cfg80211.ieee80211_regdom=DE"
Custom firstuse.sh: "#!/bin/bash\n\nset +e\n\nCURRENT_HOSTNAME=`cat /etc/hostname | tr -d \" \\t\\n\\r\"`\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom set_hostname rpissd\nelse\n   echo rpissd >/etc/hostname\n   sed -i \"s/127.0.1.1.*$CURRENT_HOSTNAME/127.0.1.1\\trpissd/g\" /etc/hosts\nfi\nFIRSTUSER=`getent passwd 1000 | cut -d: -f1`\nFIRSTUSERHOME=`getent passwd 1000 | cut -d: -f6`\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom enable_ssh\nelse\n   systemctl enable ssh\nfi\nif [ -f /usr/lib/userconf-pi/userconf ]; then\n   /usr/lib/userconf-pi/userconf 'rpissd' '$5$aSyBYXlYJy$gFt/nTEk97EB5EIvwsouaN.uO0LjvABNa84.P2O4NX4'\nelse\n   echo \"$FIRSTUSER:\"'$5$aSyBYXlYJy$gFt/nTEk97EB5EIvwsouaN.uO0LjvABNa84.P2O4NX4' | chpasswd -e\n   if [ \"$FIRSTUSER\" != \"rpissd\" ]; then\n      usermod -l \"rpissd\" \"$FIRSTUSER\"\n      usermod -m -d \"/home/rpissd\" \"rpissd\"\n      groupmod -n \"rpissd\" \"$FIRSTUSER\"\n      if grep -q \"^autologin-user=\" /etc/lightdm/lightdm.conf ; then\n         sed /etc/lightdm/lightdm.conf -i -e \"s/^autologin-user=.*/autologin-user=rpissd/\"\n      fi\n      if [ -f /etc/systemd/system/getty@tty1.service.d/autologin.conf ]; then\n         sed /etc/systemd/system/getty@tty1.service.d/autologin.conf -i -e \"s/$FIRSTUSER/rpissd/\"\n      fi\n      if [ -f /etc/sudoers.d/010_pi-nopasswd ]; then\n         sed -i \"s/^$FIRSTUSER /rpissd /\" /etc/sudoers.d/010_pi-nopasswd\n      fi\n   fi\nfi\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan '[REDACTED]' '[REDACTED]' 'DE'\nelse\ncat >/etc/wpa_supplicant/wpa_supplicant.conf <<'WPAEOF'\ncountry=DE\nctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev\nap_scan=1\n\nupdate_config=1\nnetwork={\n\tssid=\"[REDACTED]\"\n\tpsk=[REDACTED]\n}\n\nWPAEOF\n   chmod 600 /etc/wpa_supplicant/wpa_supplicant.conf\n   rfkill unblock wifi\n   for filename in /var/lib/systemd/rfkill/*:wlan ; do\n       echo 0 > $filename\n   done\nfi\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom set_keymap 'de'\n   /usr/lib/raspberrypi-sys-mods/imager_custom set_timezone 'Europe/Berlin'\nelse\n   rm -f /etc/localtime\n   echo \"Europe/Berlin\" >/etc/timezone\n   dpkg-reconfigure -f noninteractive tzdata\ncat >/etc/default/keyboard <<'KBEOF'\nXKBMODEL=\"pc105\"\nXKBLAYOUT=\"de\"\nXKBVARIANT=\"\"\nXKBOPTIONS=\"\"\n\nKBEOF\n   dpkg-reconfigure -f noninteractive keyboard-configuration\nfi\nrm -f /boot/firstrun.sh\nsed -i 's| systemd.run.*||g' /boot/cmdline.txt\nexit 0\n"
Cloudinit: "hostname: rpissd\nmanage_etc_hosts: true\npackages:\n- avahi-daemon\napt:\n  conf: |\n    Acquire {\n      Check-Date \"false\";\n    };\n\nusers:\n- name: rpissd\n  groups: users,adm,dialout,audio,netdev,video,plugdev,cdrom,games,input,gpio,spi,i2c,render,sudo\n  shell: /bin/bash\n  lock_passwd: false\n  passwd: $5$aSyBYXlYJy$gFt/nTEk97EB5EIvwsouaN.uO0LjvABNa84.P2O4NX4\n\nssh_pwauth: true\n\ntimezone: Europe/Berlin\nruncmd:\n- localectl set-x11-keymap \"de\" pc105\n- setupcon -k --force || true\n\n\n"
Custom config.txt entries: ""
Custom cmdline.txt entries: " cfg80211.ieee80211_regdom=DE"
Custom firstuse.sh: "#!/bin/bash\n\nset +e\n\nCURRENT_HOSTNAME=`cat /etc/hostname | tr -d \" \\t\\n\\r\"`\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom set_hostname rpissd\nelse\n   echo rpissd >/etc/hostname\n   sed -i \"s/127.0.1.1.*$CURRENT_HOSTNAME/127.0.1.1\\trpissd/g\" /etc/hosts\nfi\nFIRSTUSER=`getent passwd 1000 | cut -d: -f1`\nFIRSTUSERHOME=`getent passwd 1000 | cut -d: -f6`\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom enable_ssh\nelse\n   systemctl enable ssh\nfi\nif [ -f /usr/lib/userconf-pi/userconf ]; then\n   /usr/lib/userconf-pi/userconf 'rpissd' '$5$hFc3SW5chP$Y77rNzvkOsfp3MW9cYs2JZE0m6aEOszAD5EdoUG9axD'\nelse\n   echo \"$FIRSTUSER:\"'$5$hFc3SW5chP$Y77rNzvkOsfp3MW9cYs2JZE0m6aEOszAD5EdoUG9axD' | chpasswd -e\n   if [ \"$FIRSTUSER\" != \"rpissd\" ]; then\n      usermod -l \"rpissd\" \"$FIRSTUSER\"\n      usermod -m -d \"/home/rpissd\" \"rpissd\"\n      groupmod -n \"rpissd\" \"$FIRSTUSER\"\n      if grep -q \"^autologin-user=\" /etc/lightdm/lightdm.conf ; then\n         sed /etc/lightdm/lightdm.conf -i -e \"s/^autologin-user=.*/autologin-user=rpissd/\"\n      fi\n      if [ -f /etc/systemd/system/getty@tty1.service.d/autologin.conf ]; then\n         sed /etc/systemd/system/getty@tty1.service.d/autologin.conf -i -e \"s/$FIRSTUSER/rpissd/\"\n      fi\n      if [ -f /etc/sudoers.d/010_pi-nopasswd ]; then\n         sed -i \"s/^$FIRSTUSER /rpissd /\" /etc/sudoers.d/010_pi-nopasswd\n      fi\n   fi\nfi\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan 'WLAN-147081' '78599f9ef73455275df40cea7933dda78e4dfd0cc5ead1bad43259b89f4ee3a1' 'DE'\nelse\ncat >/etc/wpa_supplicant/wpa_supplicant.conf <<'WPAEOF'\ncountry=DE\nctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev\nap_scan=1\n\nupdate_config=1\nnetwork={\n\tssid=\"[REDACTED]\"\n\tpsk=[REDACTED]\n}\n\nWPAEOF\n   chmod 600 /etc/wpa_supplicant/wpa_supplicant.conf\n   rfkill unblock wifi\n   for filename in /var/lib/systemd/rfkill/*:wlan ; do\n       echo 0 > $filename\n   done\nfi\nif [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then\n   /usr/lib/raspberrypi-sys-mods/imager_custom set_keymap 'de'\n   /usr/lib/raspberrypi-sys-mods/imager_custom set_timezone 'Europe/Berlin'\nelse\n   rm -f /etc/localtime\n   echo \"Europe/Berlin\" >/etc/timezone\n   dpkg-reconfigure -f noninteractive tzdata\ncat >/etc/default/keyboard <<'KBEOF'\nXKBMODEL=\"pc105\"\nXKBLAYOUT=\"de\"\nXKBVARIANT=\"\"\nXKBOPTIONS=\"\"\n\nKBEOF\n   dpkg-reconfigure -f noninteractive keyboard-configuration\nfi\nrm -f /boot/firstrun.sh\nsed -i 's| systemd.run.*||g' /boot/cmdline.txt\nexit 0\n"
Cloudinit: "hostname: rpissd\nmanage_etc_hosts: true\npackages:\n- avahi-daemon\napt:\n  conf: |\n    Acquire {\n      Check-Date \"false\";\n    };\n\nusers:\n- name: rpissd\n  groups: users,adm,dialout,audio,netdev,video,plugdev,cdrom,games,input,gpio,spi,i2c,render,sudo\n  shell: /bin/bash\n  lock_passwd: false\n  passwd: $5$hFc3SW5chP$Y77rNzvkOsfp3MW9cYs2JZE0m6aEOszAD5EdoUG9axD\n\nssh_pwauth: true\n\ntimezone: Europe/Berlin\nruncmd:\n- localectl set-x11-keymap \"de\" pc105\n- se [... cut ...] -keymap \"de\" pc105\n- setupcon -k --force || true\n\n\n"
Unmounting: "/dev/nvme0n1"
Try to perform TRIM/DISCARD on device
BLKDISCARD successful. Discarding took 1 seconds
Zeroing out first and last MB of drive
Done zeroing out start and end of drive. Took 0 seconds
Hash of uncompressed image: "66d2bc469bc2531097039616fcf91169ffd861981e1f881030b178da1ffd64df"
Write done in 81 seconds
Verify hash: "66d2bc469bc2531097039616fcf91169ffd861981e1f881030b178da1ffd64df"
Verify done in 6.548 seconds
```

lurch, 2024-02-22:
If you actually **boot** that drive, you'll find that Raspberry Pi OS automatically resizes the root partition to use the entire disk :smiley:

tdewey-rpi, 2024-02-23:
Per @lurch's comment, @vsvjoirwv, have you booted and run through the first boot of that storage device?

vsvjoirwv, 2024-02-23:
@tdewey-rpi, yes, @lurch was absolutely right. The root partition was resized after the first boot.

closed 2024-02-23 by vsvjoirwv