raspberrypi/rpi-imager #826: [BUG]: Can't build on Linux Mint 20.3

bugjacobs, 2024-02-24:
### What happened?

I can't build the rpi-imager from source on Linux Mint 20.3 on an Intel i3-x86-64 PC

### Version

1.8.5 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

Mint 20.3 on x86-64 Intel PC.

### Selected OS

custom

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

Other

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
dh_auto_configure: error: cd obj-x86_64-linux-gnu && cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=None -DCMAKE_INSTALL_SYSCONFDIR=/etc -DCMAKE_INSTALL_LOCALSTATEDIR=/var -DCMAKE_EXPORT_NO_PACKAGE_REGISTRY=ON -DCMAKE_FIND_PACKAGE_NO_PACKAGE_REGISTRY=ON "-GUnix Makefiles" -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_AUTOGEN_VERBOSE=ON -DCMAKE_INSTALL_LIBDIR=lib/x86_64-linux-gnu ../src returned exit code 1
make: *** [debian/rules:7: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed
```

tdewey-rpi, 2024-02-26:
@bugjacobs Thanks for the report.

In this case, it looks like your configure stage has failed. Unfortunately we cannot provide support for every build environment, and as such, you will need to debug this build failure yourself. If you can identify a failure in the Raspberry Pi Imager build system, or in our build documentation, I'm happy to continue this issue or review a PR.

I would suggest you start by looking at if you can build driving CMake directly, correct the issues that identifies, and then progress to using `debuild`.

bugjacobs, 2024-02-26:
Oki thanks for the reply ! Will look into again, I've been getting better at compiling things lately ! 
So I wanted to try on this too as the flatpak takes so much space :-( And the .deb needs a newer libc or what it was.

tdewey-rpi, 2024-02-26:
@bugjacobs Just to confirm - you've found that the .deb from the releases page isn't working on Mint 20.03? Can you share the error, because I'd absolutely expect that to be operational.

bugjacobs, 2024-02-26:
Done !
[image]

tdewey-rpi, 2024-02-26:
Thanks for the update, @bugjacobs.

That's certainly not what I expected, given we aim to be compatible with Ubuntu 18.04 (glibc 2.29). I've just inspected the package for arm64, and sure enough, there's a bunch of dependencies that effectively mandate Ubuntu 22.04 or later.

@maxnet Are you building the deb on Ubuntu 22.04, by chance?

lurch, 2024-02-26:
@tdewey-rpi See also #197

bugjacobs, 2024-02-26:
This is attempted to build on Linux Mint 20.3 - x86-64, not arm !

tdewey-rpi, 2024-02-27:
Unfortunately, I'm going to have to mark this bug as 'Won't fix'.

Since Imager v1.8.1, our releases page has captured the OS requirement for Linux as Ubuntu 22.04 - so, a distro containing glibc >= 2.34.

Unfortunately, while the Qt5.5 Linux build in theory supports 18.04, there is a plan to move to Qt6 in the near future - and this change will further enforce a 22.04-class operating system must be used.

@bugjacobs I would _strongly_ recommend you upgrade your base distro if you can. While Linux Mint have extended support for 20.3 through to April 2025, our migration to Qt6 will only create further hurdles to your use of Imager.

closed 2024-02-27 by tdewey-rpi