raspberrypi/rpi-imager #827: Error downloading: SSL certificate problem: certificate has expired. [BUG]:

mrkapqa, 2024-02-26:
### What happened?

A bug happened!

tried to image for RPi2 latest raspbian image on MacOS Mojave 10.14
the image would not download instead would give an error message:

[image]

### Version

1.8.5 (Default)

### What host operating system were you using?

macOS

### Host OS Version

10.14

### Selected OS

Raspbian 32bit

### Which Raspberry Pi Device are you using?

Raspberry Pi A, B, A+, B+, and Compute Module 1

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-02-26:
Duplicate of #661 ?

tdewey-rpi, 2024-02-26:
@mrkapqa Unfortunately, as @lurch points out, this is effectively a duplicate of #661. I am therefore closing this issue for two reasons:

1. It's a duplicate of an already raised issue
2. The already raised issue contains my guidance - essentially, we require macOS 10.15 or later at this time.

closed 2024-02-26 by tdewey-rpi