raspberrypi/rpi-imager #829: [BUG]: embedded imager build throws kernal panic

oxmc, 2024-02-29:
### What happened?

When building the embedded system (buildroot) it finishes with no errors, but when booting on a pi, it starts init udev shows udevd[92]: specified group 'sgx' unknown and then throws a kernal panic

### Version

1.8.5 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

RpiOS 12 (bookworm)

### Selected OS

nothing

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

USB Pen Drive

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

oxmc, 2024-02-29:
[image]

Image of boot log

oxmc, 2024-02-29:
i even made a build from an older version of rpi-imager that i had modified, here is the boot files: https://cdn.oxmc-servers.online/uploads/my-embedded-build.zip

oxmc, 2024-02-29:
Closing, after cleaning and re-installing my os it worked, looks like it was my systems fault

maxnet, 2024-02-29:
>Closing, after cleaning and re-installing my os it worked, looks like it was my systems fault

Did you change any build settings?

As we also encountered problems after upgrading to a newer Mesa3D version for Pi 5, that it would crash if compiled optimized as size, while it did work optimized for speed (-O2).
There may be a timing specific bug out there...

oxmc, 2024-02-29:
I modified some files, played around with buildroot confit and it builds with this:

`BR2_aarch64=y
BR2_cortex_a72=y
BR2_PACKAGE_HOST_LINUX_HEADERS_CUSTOM_6_1=y
BR2_TOOLCHAIN_BUILDROOT_CXX=y
BR2_OPTIMIZE_2=y
BR2_SSP_NONE=y
BR2_FORTIFY_SOURCE_NONE=y
BR2_INIT_NONE=y
BR2_ROOTFS_DEVICE_CREATION_DYNAMIC_EUDEV=y
BR2_ROOTFS_MERGED_USR=y
BR2_TARGET_GENERIC_ROOT_PASSWD="raspberry"
BR2_ROOTFS_OVERLAY="$(BR2_EXTERNAL_IMAGER_PATH)/board/overlay/"
BR2_ROOTFS_POST_BUILD_SCRIPT="$(BR2_EXTERNAL_IMAGER_PATH)/board/post-build.sh"
BR2_LINUX_KERNEL=y
BR2_LINUX_KERNEL_CUSTOM_TARBALL=y
BR2_LINUX_KERNEL_CUSTOM_TARBALL_LOCATION="$(call github,raspberrypi,linux,2a47ccf97c6a91bc56f8cfb387d47f59cc347dd5)/linux-2a47ccf97c6a91bc56f8cfb387d47f59cc347dd5.tar.gz"
BR2_LINUX_KERNEL_USE_CUSTOM_CONFIG=y
BR2_LINUX_KERNEL_CUSTOM_CONFIG_FILE="$(BR2_EXTERNAL_IMAGER_PATH)/board/bcm2711_defconfig_minus_modules"
BR2_LINUX_KERNEL_CONFIG_FRAGMENT_FILES="$(BR2_EXTERNAL_IMAGER_PATH)/board/kernel_config_fragment_recov"
BR2_LINUX_KERNEL_IMAGEGZ=y
BR2_LINUX_KERNEL_ZSTD=y
BR2_LINUX_KERNEL_DTS_SUPPORT=y
BR2_LINUX_KERNEL_INTREE_DTS_NAME="broadcom/bcm2711-rpi-4-b broadcom/bcm2711-rpi-400 broadcom/bcm2711-rpi-cm4 broadcom/bcm2712-rpi-5-b overlays/overlay_map overlays/dwc2-overlay overlays/vc4-kms-v3d-pi4-overlay overlays/vc4-kms-v3d-pi5-overlay overlays/disable-bt-overlay overlays/disable-wifi-overlay overlays/disable-bt-pi5-overlay overlays/disable-wifi-pi5-overlay"
BR2_PACKAGE_BUSYBOX_SHOW_OTHERS=y
BR2_PACKAGE_XZ=y
BR2_PACKAGE_MESA3D=y
BR2_PACKAGE_MESA3D_GALLIUM_DRIVER_V3D=y
BR2_PACKAGE_MESA3D_OPENGL_ES=y
BR2_PACKAGE_QT5BASE_EGLFS=y
# BR2_PACKAGE_EUDEV_ENABLE_HWDB is not set
# BR2_PACKAGE_LIBOPENSSL_ENABLE_WEAK_SSL is not set
# BR2_PACKAGE_LIBOPENSSL_UNSECURE is not set
# BR2_PACKAGE_LIBOPENSSL_DYNAMIC_ENGINE is not set
# BR2_PACKAGE_LIBOPENSSL_ENABLE_COMP is not set
BR2_PACKAGE_LIBCURL_VERBOSE=y
# BR2_PACKAGE_LIBCURL_PROXY_SUPPORT is not set
# BR2_PACKAGE_LIBCURL_EXTRA_PROTOCOLS_FEATURES is not set
BR2_PACKAGE_DHCPCD=y
# BR2_PACKAGE_IFUPDOWN_SCRIPTS is not set
BR2_PACKAGE_UTIL_LINUX_BINARIES=y
BR2_TARGET_ROOTFS_CPIO=y
BR2_TARGET_ROOTFS_CPIO_ZSTD=y
BR2_PACKAGE_RPI_IMAGER=y
BR2_PACKAGE_RPI_FIRMWARE_CUSTOM=y`

oxmc, 2024-02-29:
Unsure if it was just a weird Linux issue or what

oxmc, 2024-03-08:
This seems t be happening more and more often, im going to re-open the issue

tdewey-rpi, 2024-06-19:
We have a new version of embedded Imager due imminently, and I'd love to hear if you're still seeing this, @oxmc

tdewey-rpi, 2024-06-27:
@oxmc Can you please try with the current published Embedded Imager?

oxmc, 2024-06-27:
Sure! I'll goahead and clone the new code.

tdewey-rpi, 2024-09-11:
No response in ~2 weeks, marking as stale.

closed 2024-09-11 by tdewey-rpi

oxmc, 2024-09-12:
Apologies, I have not had time to test! I’m currently writing this email on
my lunch break.