raspberrypi/rpi-imager #831: [BUG]: Existing users are overwritten when applying custom settings

besi, 2024-03-05:
### What happened?

1. I have an image with custom users pre-defined.
2. I flash the image to the an SD card using custom settings (wifi / hostname and base user)
3. The existing user was overwritten by those settings
4. The image, which relies on that deleted user is now not working as intended

### Version

1.8.5 (Default)

### What host operating system were you using?

macOS

### Host OS Version

12.6.2 (21G320)

### Selected OS

Custom Talkkonnect image

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero, Zero W, Zero WH

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-03-05:
OS Customisation should not have been offered when you selected the custom image.

We have no way of knowing, at the point of selecting a custom image, which of the customisation schemes we should use - and offering that choice to a user is not one I'm prepared to make in the GUI as I'm not convinced new users could make an informed choice at that part of the flow.

As such, I'm going to remove the OS customisation UI flow when you've selected a custom image. Note that this will _not_ remove the ability to apply customisations via the `rpi-imager --cli`.

tdewey-rpi, 2025-12-12:
I believe this is now resolved - customisation is opt-in for custom images, and at least for cloud-init should behave in the manner you expect (unless you name your user 'pi').