raspberrypi/rpi-imager #834: [BUG]: qt5 cmake error when building embedded version

CommanderGL, 2024-03-11:
### What happened?

I get the following error when I run `./build.sh`:

```
CMake Error at /home/crossscar/rpi-imager/embedded/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/lib/cmake/Qt5Core/Qt5CoreConfig.cmake:26 (message):
  The imported target "Qt5::Core" references the file

     "/home/crossscar/rpi-imager/embedded/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/include/qt5/"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/home/crossscar/rpi-imager/embedded/buildroot/output/host/aarch64-buildroot-linux-gnu/sysroot/lib/cmake/Qt5Core/Qt5CoreConfig.cmake"

  but not all the files it references.
```

### Version

1.8.5 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

Latest Arch - Kernel 6.7.9

### Selected OS

None

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

Other

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

CommanderGL, 2024-03-16:
there is no qt folder in buildroot

maxnet, 2024-03-16:
>there is no qt folder in buildroot

Probably called something like output/build/qt5base

Did you change anything in the buildroot configuration, e.g. compiler version?

CommanderGL, 2024-03-20:
I have tried rebuilding after removing qt folder and am using a fresh git clone

tdewey-rpi, 2024-06-27:
@CommanderGL Was there a particular need to rebuild the Embedded Imager? Does the current release provide what you need?