raspberrypi/rpi-imager #839: [BUG]: Wifi not working with PRI OS 32 bit lite

jedie, 2024-03-20:
### What happened?

Using the Raspberry Pi imager (v1.7.2) and setting an SSID and Password: Wifi will not working.

Think the image will create `/etc/wpa_supplicant/wpa_supplicant.conf` but Debian Bookworm will use credentials from `/etc/NetworkManager/system-connections/foobar.nmconnection`

### Version

< 1.8.0

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

Linux Mint 21.2

### Selected OS

PRI OS 32 bit lite

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero 2 W

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-03-20:
We can't fix bugs in old versions (v1.7.2) of Imager. Can you please see if v1.8.5 works better? https://github.com/raspberrypi/rpi-imager/releases/tag/v1.8.5

tdewey-rpi, 2024-04-11:
Closing this as 'Won't Fix/Stale'.

Essentially, it's unrealistic to expect a version of Raspberry Pi Imager released before Bookworm to have the knowledge of how to configure Bookworm. As @lurch has pointed out, trying the current release (v1.8.5 at time of writing), will likely resolve your concerns.

Unfortunately, however, we've gone 3 weeks without a further response - so, closing.

closed 2024-04-11 by tdewey-rpi