raspberrypi/rpi-imager #841: major accessibility issues for rpi-imager for totally blind users on all platforms using the keyboard[BUG]:

nickgaw, 2024-03-21:
### What happened?

A bug happened!I myself am totally blind and am trying to use rpi-imager to image a new SD card.  When I open up rpi-imager and try to tab threw the options I am unable to do so as tab does nothing.  Making the program fully accessible under windows and the mac both which have built in accessibility tools included would greatly help a lot.  http://www.microsoft.com/enable and http://www.apple.com/accessibility have great accessibility information and with orca in linux the gnome project has lots of information on their web site.  Windows has narrator in it and the mac has the voice over screen reader in it which lets blind users use the mac with just a keyboard.  Fighting with a mouse that I do not normally use is not the best way to use an application and the raspberry pi image release notes do talk about a way to install orca by pressing control and alt and space but this could easily be done in the imager so that on first boot accessibility is turned on automatically.  As a totally blind student who is studying programming I feel that as the raspberry pi foundation says they care about accessibility but then their tools are almost unusable with the keyboard I feel this is not the best practice and really needs to be looked into as soon as possible.

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

11

### Selected OS

the latest raspberry pi operating system

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-09-11:
Some changes in this direction have been included in v1.9.0, but I've not yet done a full walk of the application purely with keyboard due to time constraints.

I hope these changes improve the situation, and will try to get time to do the full walk within the next month.

nickgaw, 2025-07-18:
I just wanted to give an update with the latest version I am able to tab around to the devices button and the operating systems button but after I try to select a storage device and a raspberry pi model to flash the next button is not accessible with the tab key.  I even looked around the program directory to see if on windows you had command line options and it appears that none exist.  I think that using normal combo boxes or radio buttons is better then the buttons that should bring up menus of options but do not.  http://www.microsoft.com/enable and http://www.apple.com/accessibility as well as their development libraries have great accessibility information and with both windows and the mac and even some linux versions having screen reading programs built into the operating systems testing this out with just the keyboard should not be that complex.  Once I select the storage device and the operating system I should be able to tab to the next button and hit space to select it.

tdewey-rpi, 2025-12-12:
2.0 is a significant step forward in this dimension.

There may be spot-failures around specific functions, and these should be raised separately.

closed 2025-12-12 by tdewey-rpi