raspberrypi/rpi-imager #842: [BUG]: Subitems randomisation is broken

tdewey-rpi, 2024-03-25:
### What happened?

The _Home Assistants and Automation_ category should have item ordering randomisation applied - but it appears to present items in the same order on every invocation.

### Version

1.8.5 (Default)

### What host operating system were you using?

macOS

### Host OS Version

14.4

### Selected OS

N/A

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

Other

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-03-25:
Tagging @cillian64, as it looks like our V4 server side processor is dropping the random key from subitems.

cillian64, 2024-03-26:
Fixed by adding this to the schema in imager-manifester.

closed 2024-03-26 by cillian64