raspberrypi/rpi-imager #843: [BUG]: OS Customisation option is missing.

salmanfarisvp, 2024-03-27:
### What happened?

In the new version of v1.8.5, I see the OS customization option is missing in the MAC version (not sure about the other models). It was so handy to set up the network and SSH before powering the Pi. 

[image]

### Version

1.8.5 (Default)

### What host operating system were you using?

macOS

### Host OS Version

14.2.1 (23C71)

### Selected OS

Raspberry Pi OS

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-03-27:
Note that in previous versions of RPi Imager, there was a "settings-icon" on the initial screen, but in recent versions of RPi Imager it has instead moved to only be triggered as part of the flashing process - see the screenshots / walkthrough at https://www.raspberrypi.com/documentation/computers/getting-started.html#raspberry-pi-imager
Is this also what you're seeing, or is there some other way in which the OS Customisation options are "missing" for you?

salmanfarisvp, 2024-03-28:
Thanks for sharing the update update. But looks like it's not always triggering to open the settings-icon.

lurch, 2024-03-28:
Perhaps I didn't explain it clearly enough - there now **isn't** a settings-icon. Again, see the walkthrough that I linked to, to see where / when the OS Customisation happens.

jayjayseal, 2024-03-28:
ctrl+shift+x to trigger the window with settings.

closed 2024-03-28 by salmanfarisvp

varunyellina, 2025-06-02:
Date: June 2, 2025
OS: macOS 15.5
Raspberry Pi Imager: v1.8.5

It did not prompt me to customize options. I used the keyboard shortcut after see this issue.

salmanfarisvp, 2025-06-02:
Hi @varunyellina , I Just checked with Pi Image v1.8.5 and it's working fine. Please note, it will only prompt after you choose the **Pi Model**, **OS Version** and **Storage**. 

**Tested on** 
OS: macOS 15.4.1
Raspberry Pi Imager: v1.8.5

[image]

[image]