raspberrypi/rpi-imager #844: [FEATURE]:  More control over network  setup in imager

Brunnian, 2024-03-31:
### Is your feature request related to a problem? Please describe.

I want to set my devices to a static IP address,  at the moment I have to connect a screen and keyboard at initial boot, and change the IP address and gateway/dns address.

### Describe the solution you would like to see implemented

It would be very useful to be able to preset a static IP address in the imager just as the SSID and password can be set.

Similarly, it would be useful to be able to preset the VNC server to ON just as SSH login can be set.

That would mean I could create a new bootable card and connect to the device with VNC without needing the extra step with a screen and keyboard.

### Describe alternatives you've considered

I would also be happy if it was easy to modfiy these settings in text files when inspecting the flash card in a card reader, before going to the pi hardware

### Additional context

.

### Version

1.8.5 (Default)

lurch, 2024-04-02:
Duplicate of #288 ?

tdewey-rpi, 2024-04-09:
@Brunnian Feature request rejected, duplicate of #288.

I won't repeat the justification in this issue - but I would encourage you to, before submitting a feature request, search the 'Closed' issues to see if the feature has been previously rejected. I appreciate that GitHub doesn't make this obvious.

closed 2024-04-09 by tdewey-rpi