raspberrypi/rpi-imager #845: [BUG]: version 1.8.5 OS Customisation not showing/missing

Ketil64, 2024-03-31:
### What happened?

version 1.8.5 does not show any means to customize image with ssh, wifi, username and so on - reverting to older version works

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

22H2

### Selected OS

custom

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

Brunnian, 2024-04-01:
I am not having that problem.
[image]

Ketil64, 2024-04-01:
Neither did I the first time, 2nd time and so on. But yesterday after error message writing to memorycard this splash screen is not showing. Restarting app and switching memorycard reader fixed the problem, but the OS customisation was missing.
After uninstalling the app and installing 1.7.2 everything works as it should.

Been using the imager for quite som time now, perhaps the memorycard error thew something in app out of sync?

lurch, 2024-04-02:
In your initial report you mention "Selected OS: custom". IIRC RPi Imager only displays that "Use OS customisation?" popup if it detects that the OS is capable of being customised.
Can you try selecting "Raspberry Pi OS" as the OS, and see if you still fail to get the "Use OS customisation?" popup?
See https://www.raspberrypi.com/documentation/computers/getting-started.html#raspberry-pi-imager for a bunch of screenshots of the "usual workflow".

Ketil64, 2024-04-02:
in fact - I did select an image that was not capable of being customized, but when I discovered my error and selected the correct image I had no customization ability ever again. Even if I shut down the program and started all over again. The solution for me was to uninstall and downgrade to the version having the customization button.
I selected the wifi, ssh and so on with no problem.
One thing I did not do was restart my computer. So if this is a bug - I dont know - perhaps a series of mishaps.
1. Having problem with memory reader
2. changing memory reader and selecting wrong image
3. selecting correct image but unable to customize
4. restarting software, selecting correct image, but still no customization offered
5. uninstalled/downgraded and now a happy camper.
6. will upgrade to new software when image will be written

closed 2024-04-02 by Ketil64