raspberrypi/rpi-imager #849: [FEATURE]: EEPROM boot order control for embedded imager

foxt, 2024-04-15:
### Is your feature request related to a problem? Please describe.

It would be nice to have the ability to, change the boot order for the EEPROM in RPI imager. For example, if you accidentally type the order in wrong or need to enable USB/Net/SD boot. For example, if you set it to only boot from network, and don't have a network available with a net-boot server available, or the reverse, need to enable net/usb boot when you don't want to wipe an SD card.

### Describe the solution you would like to see implemented

An option to toggle the currently enabled boot modes, and even better, rearrange them.

### Describe alternatives you've considered

The same thing, but implemented in the EEPROM itsself, but with it in the Imager it would be harder to accidentally activate.

### Additional context

Pretty self explanatory 

### Version

1.8.5 (Default)

tdewey-rpi, 2024-04-22:
[image]

This is already a feature - albeit one that requires a separate write of the sdcard. Were you looking for something else?

lurch, 2024-04-22:
> need to enable net/usb boot when you don't want to wipe an SD card.

If you've already got Raspberry Pi OS on that SD card, you can change the boot-order using `raspi-config` https://www.raspberrypi.com/documentation/computers/configuration.html#advanced-options

foxt, 2024-04-22:
Hi @tdewey-rpi, Yes, ideally you would be able to change the boot order without writing an SD card (or having one in the Pi at all), however from further reading my understanding is that the EEPROM is only writable to to the boot loader?

maxnet, 2024-04-22:
It certainly do is possible.
cmprovision is able to modify eeprom setting and write with flashrom from within Linux just as well...

tdewey-rpi, 2024-04-30:
I'm not convinced this represents a great trade-off in terms of implementation effort vs wide utility.

On modern platforms, `rpiboot` can perform the function you've requested. On older platforms, there is no EEPROM.

So there's already a mechanism to do just this - and I can't see the utility in adding another. Especially when you consider that one of the use-case examples for `rpiboot` is to load the embedded imager.

As such, Feature request rejected - alternative software is already provided for this, and simple EEPROM manipulation payloads are already provided in the OS list.

closed 2024-04-30 by tdewey-rpi

maxnet, 2024-05-09:
>Especially when you consider that one of the use-case examples for rpiboot is to load the embedded imager.

But rpiboot is not necessary to load embedded imager, given you can also start it through netinstall.

Being able to change boot order through that, would be a lot more user friendly than messing with rpiboot and requiring user to learn which BOOT_ORDER=0x value has to be used...
(And while the "let Imager write a SD card with recovery.bin and a boot setting" method, does is user-friendly, it may not work with some SD cards that the ROM code does not like.)

timg236, 2024-05-13:
I agree that this isn't a high priority thing. 

However, using flashrom from the embedded image would have the advance that it could have the same behaviour as raspi-config where it only modifies the BOOT_ORDER and leaves the reset of the flash data untouched. That would require more complexity in the UI code and we probably don't want to add dependencies on other components e.g. spawning 'rpi-eeprom-config --edit' would require Python