raspberrypi/rpi-imager #852: [FEATURE]: Add locale to the imager

knutegit, 2024-04-18:
### Is your feature request related to a problem? Please describe.

Image has timezone and wifi country but defaults to en_gb.

### Describe the solution you would like to see implemented

Add a locale selection to imager

### Describe alternatives you've considered

If I remember I can use raspi-config to fix the locale.

### Additional context

No

### Version

1.8.5 (Default)

lurch, 2024-04-22:
Imager already has locale settings - see the screenshots at https://www.raspberrypi.com/documentation/computers/getting-started.html#advanced-options

Are you flashing Raspberry Pi OS to your SD card, or some other OS?

tdewey-rpi, 2024-04-30:
Feature request rejected.

This is already a feature of Raspberry Pi Imager.

scruss, 2024-05-11:
> Imager already has locale settings ...

No, it doesn't. You *can* set the time zone and the keyboard language, but `$LANG` is always set to en_GB.UTF-8 (and similarly, the output of `locale` is mostly en_GB). To have a proper locale setting, you need to have the equivalent of
```
dpkg-reconfigure locales
```

See forum thread: [Raspberry Pi Imager improvement request - Raspberry Pi Forums](https://forums.raspberrypi.com/viewtopic.php?p=2220151#p2220151)

lurch, 2024-05-11:
Oh! Sorry that I misunderstood the request.

ZuiMelanieForno, 2025-07-01:
I use this Workaround.

In the `firstrun.sh` I added the following line before the deletion-command of the firstrun.sh
`/usr/bin/raspi-config nonint do_change_locale de_DE.UTF-8`

```bash

KBEOF
    dpkg-reconfigure -f noninteractive keyboard-configuration
fi

/usr/bin/raspi-config nonint do_change_locale de_DE.UTF-8

rm -f /boot/firstrun.sh
sed -i 's| systemd.run.*||g' /boot/cmdline.txt
exit 0

```

```bash
<some prompt> locale
LANG=de_DE.UTF-8
LANGUAGE=
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=
```

Greetings