raspberrypi/rpi-imager #853: [FEATURE]: Install 1.8.5 on Ubuntu via apt

chrowe, 2024-04-21:
### Is your feature request related to a problem? Please describe.

https://www.raspberrypi.com/software/ suggests `sudo apt install rpi-imager` for installation as an option but it installs 1.7.5.

### Describe the solution you would like to see implemented

Based on https://packages.ubuntu.com/search?suite=default&section=all&arch=any&keywords=rpi-imager&searchon=names
1.8.5 is available for Ubuntu 24.04 but that is still in Beta. It would be nice to update the package version for as many older versions of Ubuntu as possible.

### Describe alternatives you've considered

Downloading the .deb file from the website give you 1.8.5
If you don't have control over the package updates, maybe just put a note on the website that it might be an older version?

### Additional context

Ubuntu 23.10 on an old i7 MacBook Pro

### Version

1.8.5 (Default)

lurch, 2024-04-21:
See e.g. #489

tdewey-rpi, 2024-04-22:
Hi @chrowe

Thanks for the submission. Unfortunately, I'm going to have to reject this feature request.

We do not publish directly to the canonical repositories - this is something that, to my understanding, @waveform80 graciously does for Ubuntu users (and this is very much appreciated!).

If you visit the Raspberry Pi Software website, you'll be directed to down
load a .deb from our own publishing mechanism - and this does indeed point at 1.8.5 today.

The website does, however, tell you that on *Raspberry Pi OS* you can use apt:
[image]

As a result, the only action I can see here to address your concern is one outside of the control of this project - in essence, your request is to @waveform80 to ask for a more recent version of Raspberry Pi Imager in the Ubuntu repositories. I'd be happy enough to consider what we can do to make this process simpler, but I cannot directly resolve your concern.

Given this, I reject the Feature Request, but empathise with your concern.

chrowe, 2024-04-22:
Do you know who manages https://www.raspberrypi.com/software/ ?
If nothing else, added a note to check the version or something. e.g.
> You can use `apt policy rpi-imager` to check the version that will be installed.

waveform80, 2024-05-02:
I'm afraid the criteria for bumping full versions of things in existing releases of Ubuntu are [pretty strict](https://wiki.ubuntu.com/StableReleaseUpdates#When) (and generally with good reason). There would typically need to be a high-impact bug fixed by the new version. Even then, I would still be asked "can you just backport the fix without bumping the version?", and following that "is using [the snap](https://snapcraft.io/rpi-imager) an option?", which realistically it is for most cases on Ubuntu.

I acknowledge the snap has some drawbacks, mostly its ridiculous size (200MB!). This is because it embeds a good chunk of Qt (because the Qt content snap is only built for amd64 and arm64, hence I can't use it and maintain support for the other archs). But it is reasonably functional, and I do try and keep it up to date with upstream.

In the meantime, noble (24.04) should have [rpi-imager 1.8.5](https://launchpad.net/ubuntu/+source/rpi-imager) and if a new version comes out before oracular's (24.10) release, I'll be sure to try and get that bumped too. But, failing actual defects, the Ubuntu deb archive is largely fixed to the versions in it at the time of its release.