raspberrypi/rpi-imager #854: [BUG]: Usage of .local as domain

besendorf, 2024-04-27:
### What happened?

When customizing  the settings during install the imager allows setting a hostname. One cannot edit the domain and it is set to ".local". .local should not be used for internal addresses as it is reserved for mDNS (https://datatracker.ietf.org/doc/html/rfc6762).
The IANA just recently reserved .internal for that purpose. 
I would suggest making the field editable and setting the default to .internal

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

-

### Selected OS

-

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-04-28:
Raspberry Pi OS doesn't actually set the "domain" part of the hostname, and it **does** use mDNS.
```console
pi5@homepi5:~ $ hostname --long
homepi5
pi5@homepi5:~ $ cat /etc/hosts
127.0.0.1	localhost
::1		localhost ip6-localhost ip6-loopback
ff02::1		ip6-allnodes
ff02::2		ip6-allrouters

127.0.1.1	homepi5
```

I believe Raspberry Pi Imager simply shows the `.local` part after the hostname, to illustrate to people what hostname they would need to use to access it (e.g. via SSH) from another computer, using the mDNS name.

tdewey-rpi, 2024-04-29:
Thanks for the report, @besendorf.

It looks like the label, which is for informational purposes only (as @lurch has pointed out), has created some confusion here. The only change I can see that would reduce the scope for confusion is to delete the label entirely - we can't know the mDNS domain used on a deployed network, so pretending we _can_ know is potentially harmful.

However, when I consider the _majority_ of deployments (indeed, literally _every single network_ I've used mDNS on), I believe the label offers a degree of useful information.

Autodetection is also inappropriate here, for the same reason as it is potentially harmful for WiFi country codes, etc - just because you're writing the SD card on a given machine; in a given locale; in a given place; does _not_ mean that the Raspberry Pi that will execute the storage device carries the same assumptions.

I can't say I'm confident of the _correct_ solution here. I'm not sure there even is one. I can, however, say that I don't intend to remove the label until we can offer a solution that both reduces confusion and improves accuracy for the majority of users.

I'll hold this open for at least 2 weeks from this day to allow for discourse - but if we cannot converge on a proposed replacement I will be forced to mark this issue as 'Won't Fix'.

cmd0s, 2024-10-18:
In my opinion, as both a user and developer, it’s best to leave it as it is. I’m aware that some routers may use suffixes other than '.local', but these are rare cases.

Removing this information would result in most users being unaware of the option to use 'hostname.local' instead of an IP address.

hortimech, 2024-12-08:
>  I’m aware that some routers may use suffixes other than '.local', but these are rare cases.

Every Plusnet router sets (via dhcp) the '.home' TLD, so it not that rare is it.

Couldn't an option be added to to set the dns domain name, with a message saying that '.local' is used if not given ?

lurch, 2024-12-08:
> Every Plusnet router sets (via dhcp) the '.home' TLD, so it not that rare is it.

Oh interesting, thanks for the heads-up!

> Couldn't an option be added to to set the dns domain name, with a message saying that '.local' is used if not given ?

As already mentioned, Raspberry Pi Imager doesn't actually **set** the  domain, it just displays the mDNS domain that _most_ users would need to use to connect to their Raspberry Pi. So if your router sets the TLD to `.home` rather than `.local`, you just need to remember to `ssh your_pi_username@your_pi_hostname.home` instead of `ssh your_pi_username@your_pi_hostname.local` .

hortimech, 2024-12-08:
Perhaps I should have mentioned that I have to turn the dhcp server on my plusnet router off because I do not want either '.local' or '.home' as a TLD.

With the setup at present, you have to use no dns domain during the install and then set the dns domain after the install has finished. I know that most users will not require a dns domain, but there are others that will.

lurch, 2024-12-08:
> Perhaps I should have mentioned that I have to turn the dhcp server on my plusnet router off because I do not want either '.local' or '.home' as a TLD.

Well, that's obviously your personal choice.

> I know that most users will not require a dns domain, but there are others that will.

...and the people that want a custom DNS domain, will have the expertise to be able to configure this themselves. Or they might even have configured their own DHCP server to automatically hand out the TLD that they _do_ want.

Raspberry Pi Imager is often the "entry point" to the Raspberry Pi ecosystem for a lot of people who have never even used Linux before; and therefore we're deliberately keeping RPi Imager easy-to-use and not overwhelming it with advanced features that 99% of people will never need to use.

tdewey-rpi, 2024-12-09:
Per my earlier comment:

> I'll hold this open for at least 2 weeks from this day to allow for discourse - but if we cannot converge on a proposed replacement I will be forced to mark this issue as 'Won't Fix'.

We're now >2 months, and I have not seen a proposal for how to resolve this in a manner that:

1. Improves accessibility of the function
2. Reduces first-time user confusion.

As I've stated numerous times through the last year - Imager is a tool aimed primarily for the newest users - those with the least technical knowledge, who will benefit greatest from a tool to simplify their initial introduction to Raspberry Pi devices. I've received many requests for more advanced features - but very rarely do these requests consider that the newest users may be entirely unaware of the concept they're requesting, and that adding their feature may simply add confusion for that user base.

I held this issue open to allow discourse and proposals - but I believe the issue has run its course. I consider this feature something that more advanced users will be able to action manually - and cannot see a way to introduce changes to Imager that would not degrade the first-user experience.

Closing as 'Won't fix'.

closed 2024-12-09 by tdewey-rpi