raspberrypi/rpi-imager #855: [FEATURE]: Static IP address

Cubeman333, 2024-04-27:
### Is your feature request related to a problem? Please describe.

it is annoying to set a static IP address on a Pi before first boot.

### Describe the solution you would like to see implemented

It would be nice if you could set a static IP address directly in the Imager.

### Describe alternatives you've considered

It would be way harder to implement a easy method of setting a static IP in the dhcp.conf, because you cant acess this partition on windows.

### Additional context

I dont know how difficult it is to do this with the different network interfaces, but it would be a nice addition to the Imager.

### Version

1.8.5 (Default)

lurch, 2024-04-28:
Duplicate of #288 (and other similar requests)

tdewey-rpi, 2024-04-29:
Closing as 'Won't Fix'.

The rationale has been documented in numerous locations now - if you're curious, a search should reveal it fairly promptly.

closed 2024-04-29 by tdewey-rpi

NickNevzorov, 2026-08-15:
I really need this function. I often write images, and then I have to run each one to set a static address