raspberrypi/rpi-imager #856: [BUG]: Verifying write failed. Contents of SD card is different then what was written to it.

kaiosod, 2024-04-28:
### What happened?

I have been struggling to flash my SD card for Raspberry OS, I have tried to use RPI imager V1.8.5, Balena Etcher and I also tried to use the USB port on the back of my computer, but the same issue keep appearing.

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Windows 11

### Selected OS

Raspberry PI OS 64 bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-04-29:
If both Raspberry Pi Imager and Balena Etcher are having problems writing to your SD card, then maybe either your SD card or your card-reader are faulty? Do you have different SD cards and card-readers you're able to try? One thing that sometimes catches people out is trying to use a new SDXC card in an old card-reader that only supports SDHC cards. https://en.wikipedia.org/wiki/SD_card#SDHC
You also might want to look at using [h2testw](https://www.softpedia.com/get/System/System-Miscellaneous/H2testw.shtml) to check that your SD card is operating correctly.

kaiosod, 2024-04-30:
Hi @lurch Thanks for feedback... I already tried to use another SD card and a different card reader but I had the same result, do you know another tool that I can use to flash the SD card ?

lurch, 2024-04-30:
Sorry to be blunt, but if neither Raspberry Pi Imager or Balena Etcher are working correctly, then I suspect any other tool would also fail to write your SD card correctly. Did you try using h2testw to perform a low-level test?
You could try using https://sourceforge.net/projects/win32diskimager/ but note that it doesn't do verification after writing, so it may _appear_ to work correctly, but give you a partially-corrupted SD card.

tdewey-rpi, 2024-04-30:
@kaiosod Ouch! This looks like you may have some severe hardware problem. If you were using a Raspberry Pi 4 or 5, I'd probably suggest trying Network install at this point - but since you're using a Raspberry Pi 3-generation platform, your options will be limited to trying the various USB ports on your computer, or possibly borrowing someone else's computer for the initial write.

Note that _after_ the initial write, you can invoke Raspberry Pi Imager inside Raspberry Pi OS, and from there write following cards using a USB card reader attached to the Raspberry Pi directly.

This isn't obviously a problem with the `rpi-imager` project itself, so I'm minded to close this issue. Closing the issue will not prevent further discussion, but it will remove it from my tracking. You may also find more advice in the Raspberry Pi Forums, where you can get a wider range of community members to chime in on suggestions.

Closing as 'Not applicable'.

kaiosod, 2024-05-01:
Hi @lurch and @tdewey-rpi , I appreciate your support, I did the SD verification, correct me I am understading wrong but looks like my SD is corrupted probably this is the reason for the issue, this is weird because this SD card is pretty new and I bought it only for flash the Raspberry PI OS, Would you guys recommend me try to recover this SD card or try a new one ?  

This is the result of H2testw verification:

Warning: Only 29987 of 29988 MByte tested.
The media is likely to be defective.
465.9 MByte OK (954176 sectors)
28.8 GByte DATA LOST (60459200 sectors)
Details:0 KByte overwritten (0 sectors)
0 KByte slightly changed (< 8 bit/sector, 0 sectors)
28.8 GByte corrupted (60459200 sectors)
0 KByte aliased memory (0 sectors)
First error at offset: 0x000000001d1e8000
Expected: 0x000000001d1e8000
Found: 0x0000000000000000
H2testw version 1.3
Writing speed: 17.7 MByte/s
Reading speed: 16.8 MByte/s
H2testw v1.4

tdewey-rpi, 2024-05-01:
Woah! That's an absolutely wild result. My initial reaction is that you may have inadvertently bought a 'fake' sd card, where they've actually sold you a card that appears to be around ~500MiB of capacity, but has been modified to claim it is a ~30GiB card.

However, in your earlier post you said you had tried _another_ card and saw the same issue - was this purchased from the same vendor/manufacturer? If you can confirm the h2testw result from that card as well you might get some confidence in what has happened. If you see consistent test results from the cards across multiple runs, you can probably be confident the issue is with the cards - if you see _variable_ results from a single card, it may be an issue with your PC or card reader.

In the event that you've managed to acquire a fake SD card, I'd encourage you to contact your retailer and try to get a card from a reputable manufacturer instead. Unfortunately I can't offer recommendations for where to do so in Brazil (indeed, I hesitate to offer recommendations outside of the Greater Cambridge Area in the UK!).

kaiosod, 2024-05-02:
I ran the verification on another SD card, this one with 8GB capacity, and encountered a similar verification result. These SD cards were purchased from different vendors. Let me try other options, and I'll keep you guys updated on the status of this final battle haha. I hope to find the root cause for this issue

Warning: Only 7672 of 7673 MByte tested.
The media is likely to be defective.
812 MByte OK (1662976 sectors)
6.6 GByte DATA LOST (14049280 sectors)
Details:0 KByte overwritten (0 sectors)
0 KByte slightly changed (< 8 bit/sector, 0 sectors)
6.6 GByte corrupted (14049280 sectors)
0 KByte aliased memory (0 sectors)
First error at offset: 0x00000000313e8000
Expected: 0x00000000313e8000
Found: 0x0000000000000000
H2testw version 1.3
Reading speed: 13.4 MByte/s
H2testw v1.4

kaiosod, 2024-05-03:
I bought a new SD card and it worked on this one, Thank you again guys for the guidance

armindogomes, 2024-07-19:
I was stuck on the same problem. Probably it was a "fake sd-card" too. I bought a new one and all worked perfectly.