raspberrypi/rpi-imager #857: [FEATURE]: Download as a zip archive

CrazyBobcatStudios, 2024-04-29:
### Is your feature request related to a problem? Please describe.

It installs to the main account on my pc which isn't mine.

### Describe the solution you would like to see implemented

A way to download as a zip archive

### Describe alternatives you've considered

Using in a vm (very impractical but works)

### Additional context

Self explanatory.
Sorry that I took 5 seconds to write this lol.

### Version

1.8.5 (Default)

ndabas, 2024-04-29:
Related to #750?

lurch, 2024-04-29:
Or perhaps it's a duplicate of #253 ?

ndabas, 2024-04-29:
Indeed, it's a duplicate of #253, but the underlying issue motivating the feature request is #750. A literal "it's a feature, not a bug."

Anyway, the Windows install story does need some attention, I can work on that if you wish, when it comes up on the roadmap.

tdewey-rpi, 2024-04-30:
As posted on #253, and #750, this is not likely to be addressed in the near term.

Additionally, the practical impact of this is increasingly small. Workarounds include using BalenaEtcher, or if you're unable to use a PC at all, using the Raspberry Pi Network Install mechanism.

Closing as 'Won't Address' for tracking - but this may appear in a later release of rpi-imager.

closed 2024-04-30 by tdewey-rpi

CrazyBobcatStudios, 2024-05-07:
Thank you for your thoughtful response 

Sorry that I took 5 seconds to write.