raspberrypi/rpi-imager #858: [FEATURE]: Add Ubuntu 24.04 LTS

maniakm, 2024-04-29:
### Is your feature request related to a problem? Please describe.

Ubuntu 24.04 LTS fix some problems that Ubuntu 23.10 have on Raspberry Pi 5

### Describe the solution you would like to see implemented

rpi-imager should list Ubuntu 24.04 LTS for Raspberry Pi 5 and Pi 4/CM4

### Describe alternatives you've considered

rpi-imager should list Ubuntu 24.04 LTS for Raspberry Pi 5

### Additional context

https://ubuntu.com/download/server
https://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/noble-preinstalled-server-arm64+raspi.img.xz

### Version

1.8.5 (Default)

lurch, 2024-04-29:
This is something that Ubuntu themselves will have to request. ping @waveform80 (who works for Canonical).

waveform80, 2024-04-29:
This is slightly bizarre. This is the second report I've had of 24.04 not showing up in rpi-imager in the last couple of days, but I [added it last week](https://git.launchpad.net/meta-release/commit/?id=5d2132dd3e555a35279d03e995348003e30e032d) straight after the release, as usual, and it shows up for me quite happily:

[image]

Does the imager cache the JSON it downloads somewhere? I didn't think it did, but perhaps something changed since the last time I looked?

If not, I guess that leaves the possibility of invisible proxy caching in user's networks? In any case, anyone encountering this: be assured that 24.04 LTS *is* in [the list](https://changelogs.ubuntu.com/raspi/os_list_imagingutility_ubuntu.json); if it's not showing up for you, it's may be worth investigating if your network is silently caching things!

XECDesign, 2024-04-29:
There's a service that runs every 5 minutes to expand [os_list_imagingutility_v3.json](https://downloads.raspberrypi.org/os_list_imagingutility_v3.json) into [os_list_imagingutility_v4.json](https://downloads.raspberrypi.org/os_list_imagingutility_v4.json). I had a look through the logs to see how reliable it is and the last time it failed to fetch the ubuntu entries was on the 18th of April and it was resolved 5 minutes later.

Unless you got very unlucky and happened to try it in that 5 minute internal, or selected the wrong pi model, it should be there.

lurch, 2024-04-29:
...and by "reloading" I believe Tom means closing RPi Imager, and then running it again :slightly_smiling_face:

tdewey-rpi, 2024-04-29:
Addressing the issue:

1. Unfortunately, this isn't a feature request against Raspberry Pi Imager, but against the OS list. This isn't the forum for such requests.
2. I cannot reproduce the report - like @waveform80, I see 24.04 LTS clearly at the top of the Ubuntu subsection. The only mechanism I can see where you might have _not_ found it either by an intrusive caching layer unique to your configuration, or because you had rpi-imager open from before the Ubuntu sublist was updated - in which case the behaviour of rpi-imager is to not re-download the OS list until you restart the application.

Given the above, I'm marking this as 'Won't Address'. If you continue to fail to see 24.04 after completely closing rpi-imager and then re-opening it, I am prepared to entertain a _Bug_ rather than a _Feature Request_.

closed 2024-04-29 by tdewey-rpi