raspberrypi/rpi-imager #861: [BUG]: "Cached on your computer" doesn't update until program restart

lurch, 2024-04-30:
### What happened?

1. Use Raspberry Pi Imager to write "Raspberry Pi OS (64-bit)" to an SD card
2. Quit and then open Imager again
3. Click on "CHOOSE OS" and "Raspberry Pi OS (64-bit)" is correctly shown as "Cached on your computer"
4. Select "Raspberry Pi OS (32-bit)" and write it to the same SD card
5. Click on "CHOOSE OS" and "Raspberry Pi OS (64-bit)" is _still_ (incorrectly) shown as "Cached on your computer" :facepalm: 
6. Quit and then open Imager again
7. Click on "CHOOSE OS" and "Raspberry Pi OS (32-bit)" is now correctly shown as "Cached on your computer"

(Additional info: I'm running Imager on a Pi5 booted from a USB3 SSD, and I'm using the Pi's microSD slot as my card-writer)

### Version

1.8.5 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

Raspberry Pi OS Bookworm

### Selected OS

Raspberry Pi OS

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-05-01:
I think I can see what's going on here. We're caching the caching indicator.

As part of reducing the number of network fetches, we _also_ avoid rebuilding the OS list unless we've made a fetch - or changed the device filter.

In this case, I reckon you probably selected 'No filtering' or 'Raspberry Pi 5', and then performed the test - in which case this is certainly errant behaviour - but also expected in the current code.

The obvious minimal-risk step is to get the OS list to rebuild after a download has completed.

lurch, 2024-05-01:
> We're caching the caching indicator.

Yeah I thought that might be the case :laughing: 

> In this case, I reckon you probably selected 'No filtering' or 'Raspberry Pi 5', and then performed the test - in which case this is certainly errant behaviour - but also expected in the current code.

Yes, I was using "No filtering".