raspberrypi/rpi-imager #862: [BUG]: Ubuntu Desktop doesn't apply/prompt for OS customization settings

MoffKalast, 2024-05-01:
### What happened?

The title pretty much covers it. When selecting any Raspbian Desktop or Lite image, there's always a prompt for OS customization, as well as all versions of Ubuntu Server. And for both versions of Ubuntu Desktop available for the Pi 5 (only tested on it so far) the prompt is skipped and the customizations are not applied at all.

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Win 10

### Selected OS

Ubuntu Desktop 24.04/23.10

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-05-01:
Closing as 'Won't Fix'.

Unfortunately, @MoffKalast, we do not directly control the customisation flag for third party operating systems. In this case, Ubuntu 24.04 has been marked explicitly as not customisable:

`            {
              "name": "Ubuntu Desktop 24.04 LTS (64-bit)",
              "description": "Desktop OS for RPi 4/400/5 models with 4Gb+",
              "icon": "https://assets.ubuntu.com/v1/85a9de76-ubuntu-icon.svg",
              "url": "http://cdimage.ubuntu.com/releases/noble/release/ubuntu-24.04-preinstalled-desktop-arm64+raspi.img.xz",
              "extract_size": 8810098688,
              "extract_sha256": "c0b6b9a0cf56b9290abae85df1fe0f184fa000d82c2492c417e08fc8da99f726",
              "image_download_size": 2729162364,
              "image_download_sha256": "016e508f44f3c82605747ad1e621710bd224df91dddf37f27903b0cd8ec3a11f",
              "release_date": "2024-04-25",
              "init_format": "none",
              "devices": [
                "pi5-64bit",
                "pi4-64bit"
              ]
            },`

I'd encourage you to contact Canonical regarding this, as they control both the OS and the JSON that feeds into Imager.

closed 2024-05-01 by tdewey-rpi

MoffKalast, 2024-05-01:
Alright, thanks for the help.

tdewey-rpi, 2024-05-02:
@MoffKalast I suspect a kind word toward @waveform80 might work wonders. Especially if they require support from my side.

waveform80, 2024-05-02:
At the moment the two sides of the Ubuntu images for the Pi have quite different methods of initial setup.

On the server side, things are simple. The server images use [cloud-init](https://cloudinit.readthedocs.io/), and support customization via rpi-imager because rpi-imager (very kindly) integrates support for cloud-init seeds written to the boot partition.

On the desktop side, things are ... in flux (he says, putting it politely). The PC desktop images now use the [new installer](https://github.com/canonical/ubuntu-desktop-provision), which also uses cloud-init to do some of its work, and which (I think?) supports some level of customization and automated installs. But the new installer doesn't support pre-installed images, like our Pi images yet, and so the Pi images aren't using it.

Instead, they're stuck using oem-config (derived from the [ubiquity project](https://launchpad.net/ubiquity)), which doesn't support customization in this fashion (via files written to the boot partition). Worse still, the Pi images now actually include cloud-init (because it's in the desktop-minimal seed, also used as the basis of the PC desktop images) ... but it's not used because oem-config is the method of setup.

Anyway, suffice it to say that not much is going to happen on this front until the new installer starts supporting pre-installed images. I'm vaguely hoping that may happen in the next cycle (24.10 ... *maybe* in time for 24.04.1 in August?), but I'll only realistically be able to do anything about this after that point. No-one wants to put work into doing this in oem-config when the plan is not to use it in future anyway. Sorry!

MoffKalast, 2024-05-03:
Ah, well, thanks for all the info. As long as the team is aware and working toward it 👍 

Honestly for my use case I'm just having some odd networking problems on desktop, without setting up something custom with systemd and iw reg the wifi locale just doesn't persist through a reboot. Somehow the WLAN country setting in the imager gets it done properly for the server and I haven't been able to figure out what exactly it does. My current plan is to hunt down that cloud-init script and apply the changes manually.

waveform80, 2024-05-03:
P.S. Feel free to select "This bug affects me" at the top of [LP: #1951586](https://bugs.launchpad.net/netplan/+bug/1951586) -- those stats do get considered for prioritization of fixes.

waveform80, 2024-05-03:
Hmm, would your network just happen to be WPA3 (SAE) only? If so, you're likely running into another known issue which [has been patched](https://github.com/raspberrypi/linux/pull/5945) recently in RaspiOS (or at least, the patch has been merged -- I don't know for sure if it's in the release kernel yet), but which I'm pretty sure hasn't made it into our kernel yet (the launchpad bug is eluding me at present, but if I can find it later, I'll post a link here).

I'm afraid the only workaround for that at present is: use a hybrid WPA2/3 setup.