raspberrypi/rpi-imager #865: [BUG]: Combo boxes in OS Customization window behave unintuitively

lennartb-, 2024-05-05:
### What happened?

All combo boxes in the OS Customization window behave "incorrectly" when compared to how they basically work in all other Windows applications:

1. There is a visible scrollbar on the side, but it's not possible to use it with the mouse. Trying to pull the scrollbar down scrolls the list upwards, which is expected when using touch controls, but this is a desktop application. I just assume that the scrollbar is just there for visuals, not for function, which is bad from an accessibility point of view. The "scrollbar" also vanishes insanely quick.
2. There is no "acceleration" when scrolling the combobox with the mousewheel. Scrolling is thus painfully slow, because apparently no matter how fast or far the mousewheel is scrolled, it scrolls the entries one by one.

Take for example the "Wireless LAN country" box. It defaults to UK, but I want to put it to DE. My first intuition is to use the mousewheel, but it takes ages to scroll up to DE, so I switch to the scrollbar. Oh, the scrollbar is not working and results in the list actually scrolling in the other direction. At least using the cursor keys behave as expected and enable me to quickly scroll up.

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

23H2

### Selected OS

Default selection, doesn't matter

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-05-07:
Thanks for the report, @lennartb-

You are correct in your hypothesis - this errant behaviour is directly linked to the specific version of the Qt library we're linking against. Raspberry Pi Imager, using QML, wound up using widgets that were designed for a touch-first UI.

As such, for the scrolling, for example, you might get more success by 'dragging' the list with your primary mouse button.

I'm expecting this to be at least partially resolved by #818, however I'm not able to dedicate the required time to resolve the various UI issues that upgrade presents in the near future.

I'm going to hold this one open to call out the specific behaviours you've mentioned, to confirm the Qt upgrade will resolve this.

tdewey-rpi, 2024-07-17:
I can confirm I expect this to be resolved in the upcoming v1.9.0, where in my internal testing I've confirmed that:

- ComboBoxes will let you type the destination, with autocomplete
- Scrolling works as expected
- The touch-style drag-and-fling mechanism still works

tdewey-rpi, 2024-09-12:
@lennartb- If you have the time, I'd appreciate your take on the behaviour of v1.9.0 - available in the 'Releases' area of this repository.

lennartb-, 2024-09-12:
@tdewey-rpi Sure! I've tried to replicate my initial use case and did some smoke testing:

- Autocompletion with manually typing works
- Scrolling with the mousewhell is still pretty slow, but seems faster than in 1.8.5
- Scrollbars in Comboboxes still don't work and are visual only

I don't have any other OS available to test atm, is this different under macOS or Linux?

A bit unrelated:

- Selecting text via mouse doesn't work reliably. For example, click in the SSID name textbox and try to select the text via click-and-drag - most of the time that doesn't work (sometimes it does, not sure what the conditions are though). Double-clicking works, as does CTRL-A and Alt+cursor keys. It does seem to work after e.g. moving left or right with the cursor keys in the textbox.

tdewey-rpi, 2024-09-13:
> @tdewey-rpi Sure! I've tried to replicate my initial use case and did some smoke testing:
> 
> * Autocompletion with manually typing works

I can confirm this also works on macOS - and I expect it to also work on Linux.

> * Scrolling with the mousewheel is still pretty slow, but seems faster than in 1.8.5

I've noticed this as well - unfortunately there's still no obvious mechanism to increase the scroll speed. Rather than treating the items as a canvas, it looks like it's mapping 'clicks' of a scrollwheel to position increment/decrement, which produces the lumpy scrolling behaviour.

> * Scrollbars in Comboboxes still don't work and are visual only

I can confirm I can reproduce this. I can't see a way to change that behaviour without doing some intrusive customisation of the ComboBox class - which may have to wait for v2.0.0

> 
> I don't have any other OS available to test atm, is this different under macOS or Linux?

Thankfully, this appears to be pretty consistent, which meets my objectives for this release.

> A bit unrelated:
>
> * Selecting text via mouse doesn't work reliably. For example, click in the SSID name textbox and try to select the text via click-and-drag - most of the time that doesn't work [...]

I can reproduce that, and it's very much unexpected - we tell those controls to allow selection by mouse, and I agree that I would have expected that to operate without using keyboard inputs.

Thanks for taking a look. I'm going to introduce a change in v1.9.1 to allow selection by mouse in the ComboBoxes too, but I think we might be due a new bug for the specific cases we've seen.