raspberrypi/rpi-imager #868: [BUG]: Imager installation configuration

WhoIsAvailable, 2024-05-10:
### What happened?

A bug happened! 
When the user chooses to set the installation configuration options, the Imager should check to see whether SSH is actually available on the host machine before allowing the selection of the generation of a new private key. If SSH is not available then the Imager should not allow the selection of that option.

It should also explain what information is required to be entered on the line underneath the “Allow” button.

The “Getting started - Raspberry Pi Documentation” also needs to provide more information about the use of this option.

I was running on Win7Pro SP1 where it seems that I have no native version of SSH.

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Win7Pro SP1

### Selected OS

Bookworm 64 bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

USB Pen Drive

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
None whatsoever
```

lurch, 2024-05-12:
> I was running on Win7Pro SP1

I strongly suspect that we won't be making tweaks to Imager to support an OS that's been [officially unsupported](https://en.wikipedia.org/wiki/Windows_7#End_of_support_(after_January_14,_2020)) for over 4 years.

> The “Getting started - Raspberry Pi Documentation” also needs to provide more information about the use of this option.

If you have documentation suggestions, please create a separate issue at https://github.com/raspberrypi/documentation

WhoIsAvailable, 2024-05-25:
As reported, the bug has nothing to do with the fact that I am still running Win7. I could agree with your decision if you provided evidence that the Imager does check for the existence of SSH or some similar product when running under other operating systems.

maxnet, 2024-05-25:
It does check for the existence, and grays out the "ssh key-gen" button for you.
It still allows you to check the "use public authentication" box so that you can manually copy and paste a public key, you generated though other means though.

Why do you believe that is a problem?

WhoIsAvailable, 2024-05-25:
In that case, I can understand why you do not feel that it would be worthwhile making a change just for Win7. Thanks for the clarification.

tdewey-rpi, 2024-09-11:
Marking as 'Won't Fix'

Imager does check for the existence of SSH on Windows, but this check only operates on Windows 10 or later. Windows 10 is now rapidly approaching End of Life - meaning Windows 7 is _very, very end-of-life_, and I cannot reasonably make efforts to support OSs not receiving support from their vendor.

closed 2024-09-11 by tdewey-rpi