raspberrypi/rpi-imager #869: [BUG]: SSID containing a SPACE character will not connect

KGolomb, 2024-05-12:
### What happened?

Trying to configure wifi for a headless Pi Zero in the Imager application.  Any SSID that contains the space character does NOT connect successfully.  The Bookworm LITE and Bullseye LITE releases were attempted.  Connecting to an SSID without a space character does seem to work successfully.  I eventually had to flash a Bullseye image with belenaEtcher and use a wpa_supplicant.conf file in order to get wifi to successfully connect.

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Windows 10

### Selected OS

Bookworm Lite / Legacy Lite

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero, Zero W, Zero WH

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
None
```

mateossh, 2024-07-30:
I couldn't reproduce this issue on macOS 14.4. I flashed both Bookworm Lite and Legacy Lite for Raspberry Pi Zero W and the device connected to a SSID with a space without any issues.

tdewey-rpi, 2024-09-13:
Thanks for the report, @KGolomb 

It's entirely possible this is a Windows vs macOS bug, as @mateossh didn't encounter it on macOS 14.4.

I'd be very interested to know if you encounter it again with Raspberry Pi Imager v1.9.0 - which uses a much newer version of the UI library, so may have changed behaviour.

closed 2025-12-12 by tdewey-rpi