raspberrypi/rpi-imager #873: [BUG]: MacOS crash during flash

hopskipnfall, 2024-05-22:
### What happened?

I've tried this a few times and can consistently crash my mac by doing the following:

1. Take my MicroSD card (which came with my pi ver 1) and put it into the SD card adapter.
2. Put the SD card adapter into my macbook pro's SD card slot
3. Flash the device using the settings below

It takes a few minutes to flash, and after the progressbar goes to 100% it goes back to 0% and the text changes to "確認" (I assume this would be "Verifying" in English). After this starts the mac immediately crashes.

### Version

1.8.5 (Default)

### What host operating system were you using?

macOS

### Host OS Version

14.4

### Selected OS

Raspberry Pi OS Lite (32 bit)

### Which Raspberry Pi Device are you using?

Raspberry Pi A, B, A+, B+, and Compute Module 1

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
Logs from Mac report window appearing after startup post-crash:

panic(cpu 0 caller 0xfffffe002b49ccf8): "dart-apcie1 (0xfffffe20002a8000): DART(DART) error: SID 1 PTE invalid exception on write of DVA 0x543d000 (TTBR 0 SEG 0x2 PTE 0x50f) ERROR_STATUS 0x81000404 TIME 0x80034650d6 TTE 0 AXI_ID 0" @AppleT6000DART.cpp:1407
Debugger message: panic
Memory ID: 0x6
OS release type: User
OS version: 23E214
Kernel version: Darwin Kernel Version 23.4.0: Wed Feb 21 21:44:43 PST 2024; root:xnu-10063.101.15~2/RELEASE_ARM64_T6000
iBoot version: iBoot-10151.101.3
secure boot?: YES
roots installed: 0
Paniclog version: 14
KernelCache slide: 0x0000000021648000
KernelCache base:  0xfffffe002864c000
Kernel slide:      0x0000000021650000
Kernel text base:  0xfffffe0028654000
Kernel text exec slide: 0x0000000022b84000
Kernel text exec base:  0xfffffe0029b88000
mach_absolute_time: 0x3128d9ca7
Epoch Time:        sec       usec
  Boot    : 0x664ca387 0x0002bcf3
  Sleep   : 0x00000000 0x00000000
  Wake    : 0x00000000 0x00000000
  Calendar: 0x664ca5a6 0x0005c0f1

Zone info:
  Zone map: 0xfffffe1000364000 - 0xfffffe3000364000
  . VM    : 0xfffffe1000364000 - 0xfffffe14cd030000
  . RO    : 0xfffffe14cd030000 - 0xfffffe16669c8000
[... 286 lines cut ...]
com.apple.iokit.IOHDCPFamily	1.0.0
com.apple.iokit.IOCECFamily	1
com.apple.iokit.IOAudio2Family	1.0
com.apple.driver.AppleIISController	340.1
com.apple.driver.AppleAudioClockLibs	340.8
com.apple.driver.FairPlayIOKit	71.10.0
com.apple.driver.AppleARMPlatform	1.0.2
com.apple.iokit.IOSlowAdaptiveClockingFamily	1.0.0
com.apple.iokit.IOReportFamily	47
com.apple.security.quarantine	4
com.apple.security.sandbox	300.0
com.apple.iokit.IOStorageFamily	2.1
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.iokit.CoreAnalyticsFamily	1
com.apple.security.AppleImage4	6.3.0
com.apple.kext.CoreTrust	1
com.apple.iokit.IOCryptoAcceleratorFamily	1.0.1
com.apple.kec.pthread	1
com.apple.kec.Libm	1
com.apple.kec.Compression	1.0
com.apple.kec.corecrypto	14.0

** Stackshot Succeeded ** Bytes Traced 356153 (Uncompressed 890368) **
```

tdewey-rpi, 2024-05-22:
I'm not convinced this is directly a bug in Imager - we shouldn't have the power to bring down the hybrid-kernel design of macOS. I suspect the smoking gun is in the early parts of the report:

`SID 1 PTE invalid exception on write of DVA 0x543d000 (TTBR 0 SEG 0x2 PTE 0x50f) ERROR_STATUS 0x81000404 TIME 0x80034650d6 TTE 0 AXI_ID 0" @AppleT6000DART.cpp:1407`

Unfortunately I don't have enough knowledge of Apple's SoC to tell you precisely what went wrong, but it appears to be reporting an invalid page table entry while trying to write to a virtual address - all of which would be managed by Apple's kernel drivers, specifically: `AppleT6000DART.cpp:1407`

Assuming you were comfortable with the information disclosure, I hope you reported the problem to Apple. Can you reproduce this issue using Imager?

hopskipnfall, 2024-05-22:
Hi Tom, thanks for the response.

Agreed.. I was surprised this was something that could even be caused by Imager, but at the very least by following those steps outlined above using the Imager app I can consistently crash my mbp in the same way at the same place in the flashing process. I just thought I'd report it here in case it's a pattern for other users.

tdewey-rpi, 2024-05-22:
I'll try and reproduce it on my M1 Pro.

tdewey-rpi, 2024-05-22:
Tried to reproduce this on my Mac running 14.4.1 and Imager 1.8.5 and failed.

This isn't entirely surprising, of course - I use Imager on a daily basis and have never seen anything quite like the report you've found. Nor would I expect it to be so reproducible.

Do you also see it with other SDcards, or with other writing applications (eg, Balena Etcher)?

lurch, 2024-05-31:
Looks like users of Balena Etcher on macOS have been getting the same error too
https://github.com/balena-io/etcher/issues/4215

tdewey-rpi, 2024-06-27:
Based on that report from Balena, which implicates the same Apple kernel driver, I'm inclined to close this as `Won't fix`, because there's nothing inside Raspberry Pi Imager to change.

Unfortunately I'm also not aware of a public Apple Radar issue for this problem, so I'd encourage @hopskipnfall to contact Apple for an update.

closed 2024-06-27 by tdewey-rpi

Chizkiyahu, 2024-07-18:
I have the same error 3 times today during writing or validation 
reinstall the imager solve that