raspberrypi/rpi-imager #874: [QUESTION]/[FEATURE]:  Is there a way to use all devices instead of just external ones?

TK1987, 2024-05-22:
### Is your feature request related to a problem? Please describe.

With built-in hard disks, it is often inconvenient to have to remove them first in order to install the system.
It is usually easier to boot the pi via usb device and roll out an image, e.g. via ssh. 

Unfortunately, I have not yet found a way under Linux to select other devices such as loop devices as a target in order to create a customized image with the rpi-imager. 

### Describe the solution you would like to see implemented

A way to disable filtering for external devices, or a way to save a customized image with the RPI imager.

### Describe alternatives you've considered

Write an image with the customizations to an external device, create an image from it and then distribute it on the device. However, this is very cumbersome.

### Additional context

no

### Version

1.8.5 (Default)

tdewey-rpi, 2024-05-24:
> Unfortunately, I have not yet found a way under Linux to select other devices such as loop devices as a target in order to create a customized image with the rpi-imager.

I fear you're looking at this the wrong way. rpi-imager is not designed for mass provisioning - other tools are available today for certain hardware units (eg, cm-provisioner), with even more in development.

For creating pre-customised images or creating mass-deployment images, I must strongly encourage you look at pi-gen, the tool we use to create Raspberry Pi OS. This will prove a much more durable mechanism than relying on OS Customisation in a loopback manner.

If I've misunderstood the use case, please let me know - but it does look like something orthogonal to `rpi-imager`, which is strictly focussed on helping new users create bootable media for their Raspberry Pi devices.

TK1987, 2024-05-27:
> I fear you're looking at this the wrong way. rpi-imager is not designed for mass provisioning - other tools are available today for certain hardware units (eg, cm-provisioner), with even more in development.
It is not about mass provision. 

Nowadays, the Pi is no longer necessarily operated via an SD card. The Pi 5 even has a PCIe 2.0 x1 interface. 
So what if someone (let's say a beginner) wants to install the system on an M.2 NVME hard disk? To be able to do this with the RPI imager, you need either an M.2 docking station or an external case - even an internal M.2 port in a desktop computer could not be used for this, as the RPI imager always filters for external disks. 
And even if you had one, you would always have to remove the hard disk from the Pi first.

Why not just add an option to disable the filter for external devices? Something like “Show all devices"

lurch, 2024-05-28:
Or you could use the network-boot-RPi-Imager feature :slightly_smiling_face: 
https://www.raspberrypi.com/documentation/computers/getting-started.html#install-over-the-network

TK1987, 2024-05-28:
> Or you could use the network-boot-RPi-Imager feature 🙂 https://www.raspberrypi.com/documentation/computers/getting-started.html#install-over-the-network

That would of course be a possibility. But this would require at least a monitor and a keyboard. 

In my opinion, a very simple option to deactivate the device filter would give the user significantly more leeway in the various use cases.

Anyway, since this does not seem to be accepted here, I will close.

lurch, 2024-05-28:
> In my opinion, a very simple option to deactivate the device filter would give the user significantly more leeway in the various use cases.

...and also make it significantly easier for a less-experienced user to shoot themselves in the foot and e.g. accidentally overwrite their Windows installation with a copy of Raspberry Pi OS.

TK1987, 2024-05-28:
> ...and also make it significantly easier for a less-experienced user to shoot themselves in the foot and e.g. accidentally overwrite their Windows installation with a copy of Raspberry Pi OS.

A user who willfully goes into the options, selects "Show all devices" and selects his system hard disk without any sense or reason is only to blame.

van0014, 2025-04-16:
> A user who willfully goes into the options, selects "Show all devices" and selects his system hard disk without any sense or reason is only to blame.

This is exactly the way I see it. I want to be able to install Pi to my NVME, and not have to pay for an external reader, when my motherboard already handles this. My PC motherboard can fit several NVME disks, but because of a software lockout, it's useless to me. Many programs throw up an explicit warning if you're trying to use an internal disk, and that would be appropriate here. Or a simple check for a windows operationg system, because linux users generally will already know. Although it'd be good to check for both, if a big warning popup won't do