raspberrypi/rpi-imager #877: [FEATURE]: Add Talos Linux

rothgar, 2024-06-03:
### Is your feature request related to a problem? Please describe.

I'm new to this project and would like to add [Talos Linux](https://talos.dev) to the list of specific-purpose OSs. Is there documentation for how I can add a new OS? I'm sorry if there were docs somewhere for how this works. I didn't see any in the readme or official site.

### Describe the solution you would like to see implemented

Talos Linux should be an option for Raspberry Pi 4b models and more in the future when u-boot support is added.

### Describe alternatives you've considered

Currently people download an image from https://factory.talos.dev/ and manually flash it to a SD card or USB drive.

### Additional context

I'm happy to PR the changes if someone could help me find the correct place to do it.

Thank you

### Version

1.8.5 (Default)

lurch, 2024-06-03:
See https://github.com/raspberrypi/rpi-imager/issues/870#issuecomment-2125238407

tdewey-rpi, 2024-06-05:
@lurch has provided the link, but for my curiosity - why are you dependent on U-Boot?

rothgar, 2024-06-05:
I don't actually know the details for how uboot or the rpi boot system works. From what I understand since the pi doesn't have a BIOS/UEFI we need uboot to load the Linux kernel. It is what we have been using in Talos on multiple SBCs but if there's an alternative way to support more devices with a general Linux LTS kernel I'd be interested to learn more.

rothgar, 2024-06-05:
Can you help me understand a couple things about the json fields?

1. Is there a way to automate the release? Will the imager automatically pull a latest version of the json and image for each release?
2. Is there a list of supported devices? I see pi4-32bit does that also mean there's a pi4-64bit?
3. What is the init_format field? Talos is completely API driven and has a custom PID 1 init system.

Besides https://rpi-imager-stats.raspberrypi.com/ is there a way to get additional stats about which devices are installing the OS?

Thanks

rothgar, 2024-06-05:
I looked through the example for question 2 in the json schema and found some devices listed so I think I have it set correctly in my json https://github.com/rothgar/talos-rpi/blob/main/talos.json

I filled out the Google form.

tdewey-rpi, 2024-06-06:
> I don't actually know the details for how uboot or the rpi boot system works. From what I understand since the pi doesn't have a BIOS/UEFI we need uboot to load the Linux kernel. It is what we have [...]

The Raspberry Pi Bootloader can load Linux directly - consult the official documentation on building your own kernel for an example of how to set that up. Should make your deployment a lot simpler.

>Is there a way to automate the release? Will the imager automatically pull a latest version of the json and image for each release?

We automatically ingest your supplied subitems JSON file on a 5 minute timer.

>Is there a list of supported devices? I see pi4-32bit does that also mean there's a pi4-64bit?

The list is in the combined os_list, http://downloads.raspberrypi.com/os_list_imagingutility_v4.json

>What is the init_format field? Talos is completely API driven and has a custom PID 1 init system.

Not required in your case - it's a slightly awkward flag for specifying if you support OS customisation through either Cloudinit or a custom systemd service.

> Besides https://rpi-imager-stats.raspberrypi.com/ is there a way to get additional stats about which devices are installing the OS?

There are no further statistics at this time - if you need more data, I'd encourage you to implement something in your OS.

Closing this issue, I'll follow up on the form by e-mail.