raspberrypi/rpi-imager #878: Raspberry Pi Imager v1.8.5 replaces username password and wifi password with hash

CitrusBobiel, 2024-06-05:
### What happened?

After entering my desired username and password and wifi network and password, selecting enable ssh, I run the imager and start up my Rpi. I do have a touchscreen , luckily, this is for a 3D printer, so I can see that the wifi password has been changed and I cannot ssh into the Rpi because that password has been changed too.

How can I use the imager so the passwords stay as entered and I can use my Rpi headless? 

Rpi4
64bit OS

### Version

1.8.5 (Default)

### What host operating system were you using?

macOS

### Host OS Version

Ventura 13.6.4

### Selected OS

Raspberry Pi OS A Port of Debian Bookworm (64bit)

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-06-07:
Thanks for the report, @CitrusBobiel.

Unfortunately the hashing is not likely to be the source of your problem. `rpi-imager` does not support using the passwords in a plaintext manner, instead pre-hashing them in a manner that is acceptable to the software that will consume them. In the case of failure, aside from a report of damaged hardware, I have only seen instances where this has been the result of an error entering the password.

This particular issue has been discussed numerous times on this issue tracker - and I have yet to receive a compelling argument to remove the hashing behaviour. I'm happy to accept that I might be wrong - but on this issue I don't see much more scope for new arguments for or against.

Closing as 'Won't Fix', because the hashing is expected.

closed 2024-06-07 by tdewey-rpi

CitrusBobiel, 2024-06-07:
I'm confused as to what you mean. If I enter a password in the imager and the password changes when I try to access the Rpi, how is that not an issue?

tdewey-rpi, 2024-06-07:
The password you entered in Imager underwent a one-way transformation into a form that the software on the Raspberry Pi can use. That transformation is not designed to be reversible. If it were, then anyone provisioning a Raspberry Pi would gain two pieces of knowledge (in the case of the WiFi password, for example):

1. How to access the WiFi network
2. A credential they could use for stuffing attacks against other systems

The one-way hash eliminates (2), but cannot address (1).

That you see a hash when you open the file manually is by design, and explicitly designed to avoid leaking (2).

CitrusBobiel, 2024-06-07:
Ok, so what is the procedure to access the Rpi if the passwords are different than what I entered in the imager? If I can't ssh into the Rpi with my passwords I can't use it.

tdewey-rpi, 2024-06-07:
> Ok, so what is the procedure to access the Rpi if the passwords are different than what I entered in the imager? If I can't ssh into the Rpi with my passwords I can't use it.

The passwords are not meaningfully changed. Use the same password you entered into Imager.

Attempting to view the files directly will not show the password as they have been pre-hashed - which the tools would have done anyway as part of authentication.

CitrusBobiel, 2024-06-07:
Thank you. I now understand that the visual representation of the passwords are not the same.

Entered ****** and it looks like *********************************** on the Rpi.

Now the problem is the passwords that were entered are not working. I am entering the passwords correctly but they do not work. The Rpi will not log into the Wifi account with password entered in the imager. I select a non password wifi so I can connect and when I ssh and am presented with the password request. The user password does not work. 

I've re imaged 14 times atleast and the same issue happens. I deleted and reinstalled the imager and still the same. One time on the second try everything worked but there was another issue not related to this and I had to start again. 

I've noticed similar experience in different only forums and have never seen clear solution. I've followed the very simple and clear steps in the imager and nothing seems to work as it should. 

What could possibly be going on that this is not working for me?

lurch, 2024-06-07:
> One time on the second try everything worked...

That indicates that the problem here isn't with RPi Imager, so you should probably ask for help at https://forums.raspberrypi.com/

CitrusBobiel, 2024-06-07:
A 7% chance of the imager working is not a good track record. 

Actually the one time that worked was with the imager before the update. So I think there is something wrong with the imager. So 100% failure on this current version.

Once again the imager does not transfer the data entered in the setup correctly.

When I create work around I will post it here for all of the others having this same problem. 

The forum is of no help, it's just hours of wasted round and round when the solution is to fix the product or improve documentation.

Kbottoms, 2024-06-09:
This is a widely reported issue with the V1.8.5 of the imager.

The cause identified by community forums and I've personally confirmed is that regardless of localization settings, the Rpi imager will set them to GB even if you set custom settings with correct localization. Those settings are not saved. They keyboard layout differences will cause and capital letters and special characters to be incorrect.

I've repeatedly tested through re-imaging and ensuring there's no differences from the password (such as copying from a text doc) but still the password, wifi country, wifi password and keyboard layout will always set to GB and any password with capital letters or special characters will not work.

The work around used has been to set the password to something extremely basic with no uppercase or special characters then changing the localization to be the correct one.

Excuse my language but this latest imager version has been a hot mess.

tdewey-rpi, 2024-06-10:
This has now strayed off topic, and as such I'm going to lock the issue.

In summary:

* I don't believe this is a localisation issue immediately. I've seen numerous issues in this area, and they broadly fell into three major buckets:
  * Qt's QML Cache bug, resolved earlier in the 1.8 series
  * Our Qt TextInput field clearing behaviour was broken. Also resolved earlier in the 1.8 series.
  * User error
* The second reported issue (keydata not being correctly passed through) apparently affects both SSH _and_ WiFi credentials, which is deeply surprising as the WiFi standards specify a PSK is a 64 octet entity, and not a string. It's hard to see how locale _could_ affect such a thing even if it wanted to, without multiple components ignoring the spec.

Unfortunately, however, rather than having separate trackable bugs for specific behaviours, this has become a dumping-ground for pet issues. This is helpful neither to myself, nor to anyone trying to find a resolution at a later date.

As ever, this project is prepared to accept PRs, if @Kbottoms or @CitrusBobiel find themselves sufficiently motivated. Otherwise, as I get time to look at Imager I will continue to try to improve matters.