raspberrypi/rpi-imager #880: [BUG]: Devices list and OS list totally unreadable in rpi-imager booted from network

mimmus, 2024-06-08:
### What happened?

After upgrading EPROM to latest version, I tried to activate "rpi-imager" at boot, pressing SHIFT key, It downloads from the Internet and starts correctly but "Devices list" and "OS lists" are totally unreadable on a 43" TV connected by HDMI (very very very small characters).

### Version

1.8.5 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

At boot

### Selected OS

Custom

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-06-10:
I'm guessing it must be a 4K TV, and embedded RPi Imager isn't scaling up the fonts to a larger size?
@tdewey-rpi / @maxnet as a temporary workaround, is there any way of forcing embedded-imager to e.g. run in 1080p mode even if the display is 4K?

tdewey-rpi, 2024-06-10:
This is indeed a known issue - a newer build of the Embedded Imager is certainly required, but it's unclear if that will be enough.

It's a good suggestion, @lurch - rather than having the system select the highest-resolution mode, selecting a lower-resolution mode. We may, however, run into issues around non-native modes on user displays going a little bit weird.

The preferred solution is to have Qt identify the size of the display and use it's HiDPI support directly. It's not clear to me that Qt5 will do that - it may be that the Embedded Imager requires Qt6 for the preferred solution to operate correctly.

maxnet, 2024-06-10:
>I'm guessing it must be a 4K TV, and embedded RPi Imager isn't scaling up the fonts to a larger size?
>@tdewey-rpi / @maxnet as a temporary workaround, is there any way of forcing embedded-imager to e.g. run in 1080p mode even if the display is 4K?

There were two known issues I left to Peter/Tom to fix, when I threw the patch to update embedded Imager for Pi 5 over the fence in october. :-)

- upgrading to the new Mesa and/or changing fkms to kms broke the existing functionality we had to scale up things a little larger.
Commented out the QT_SCALE_FACTOR stuff here: https://github.com/raspberrypi/rpi-imager/blob/qml/src/main.cpp#L89 as it wasn't doing anything anymore.
Changing resolution instead of scaling may also be possible, but is not the cleanest solution.

- With the new Mesa, if you compile "optimized for size" (which we used to do before) it crashes on launching the application.
Shoved that issue under the carpet by compiling -O2 instead. But there is likely still a race condition somewhere that crashes things depending on timing, that really should be addressed...

Don't think either got fixed.

mimmus, 2024-06-10:
> I'm guessing it must be a 4K TV, and embedded RPi Imager isn't scaling up the fonts to a larger size? 
Yes, exactly. Really unreadable also using a magnifying glass!

mimmus, 2024-09-13:
is this solved in 1.9?

tdewey-rpi, 2024-09-13:
1.9 isn't deployed in the Network Install image at this time - I'm looking at repackaging it to use an AppImage.

thatguywiththekids, 2024-11-16:
The text is also unreadable on Sony KDL-40E4000, and that is not 4K.

lurch, 2024-11-16:
> The text is also unreadable on Sony KDL-40E4000, and that is not 4K.

I did a google and found https://www.sony.co.uk/electronics/support/televisions-projectors-lcd-tvs/kdl-40e4000/specifications which I wouldn't expect to have any problems. Could you please share a photo of how the text is "unreadable"?

tdewey-rpi, 2024-11-18:
As per @lurch's request, for anything where the symptom is a visual effect, a photo really _does_ speak a thousand words - and it greatly appreciated. Without one, I'm constrained by the hardware I have on my desk.

thatguywiththekids, 2024-11-18:
This looks fine:
[image]

This is too small to read:
[image]

lurch, 2024-11-19:
I've just tested network-boot RPi Imager (a.k.a. embedded Imager) on a Pi 5 with the 24 inch 1080p monitor on my desk, and it looks totally fine.
I wonder if your monitor is reporting some weird DPI setting in its EDID, which is confusing Qt in the embedded Imager?