raspberrypi/rpi-imager #881: [FEATURE]:  Subitems schema element for link to Setup Guide

moodeaudio, 2024-06-10:
### Is your feature request related to a problem? Please describe.

When user views list of OS choices there is no link to a Setup Guide that can provide user with required steps to prep the image for example enabling SSH, specifying userid and password, etc. As a result user writes image w/o knowing to do the required steps and then moode app does not work correctly and a support ticked is sent to our Forum.

If user downloads the image from our Website this does not occur because its stated to first read the Setup Guide.

### Describe the solution you would like to see implemented

Schema element for example "setup_guide" consisting of Title,URL

"url": "https://github.com/moode-player/moode/releases/download/r901prod/image_2024-05-30-moode-r901-arm64-lite.zip",
"extract_size": 4781506560,
"extract_sha256": "2597dabd751641127934892640b76f75305047a8cd094ba9f89197c5371ad3e1",
"image_download_size": 1350044374,
"description": "Audiophile streamer for the Raspberry Pi",
"icon": "https://moodeaudio.org/assets/img/logo/moode-logo-d.png",
"name": "moOde audio player 9.0.1 (64-bit Bookworm)",
"release_date": "2024-05-30",
**"setup_guide": "Setup Guide,https://github.com/moode-player/moode/blob/develop/www/setup.txt",**
"devices": ["pi5-64bit", "pi4-64bit", "pi3-64bit"],
"init_format": "systemd"

### Describe alternatives you've considered

None

### Additional context

None

### Version

1.8.5 (Default)

lurch, 2024-06-10:
Duplicate of #751 ?

moodeaudio, 2024-06-10:
Wow, indeed it is. I lost track of that one.
How to know whether that schema element is available yet?

lurch, 2024-06-10:
@tdewey-rpi will probably be able to confirm for definite tomorrow, but I don't believe that there have been any schema changes since you opened that previous issue.

moodeaudio, 2024-06-10:
Ok thx, I'll close this since its a dup.

closed 2024-06-10 by moodeaudio