raspberrypi/rpi-imager #885: [BUG]: Pi 5 Kali Image Download Corrupt

sam-david, 2024-06-19:
### What happened?

I have tried on three separate machines now with fresh Raspberry Pi imager installs and separate sd cards to download and write the Kali image for Raspberry Pi 5. Each time I get a "Download corrupt. Hash does not match" error. I noticed this happened sometime around downloading the latest version but not sure if that is related. Perhaps it has to do with the image being pulled down itself?

[image]

### Version

1.8.5 (Default)

### What host operating system were you using?

macOS

### Host OS Version

Sonoma 14.3.1

### Selected OS

Kali Linux (Raspberry Pi 5 - 64-bit)

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-06-19:
I can reproduce this, and will escalate to my point of contact with Kali Linux.

Unfortunately, as this is not an issue with `rpi-imager` itself, I must close this report - but I'll drop another comment once I've had notice from the KL developers they're back in shape.

closed 2024-06-19 by tdewey-rpi

sam-david, 2024-06-19:
Thanks @tdewey-rpi! I was able to reproduce this on my windows 10 machine with 1.8.5 and even a downgraded version of 1.8.4 so I figured it was the image as well. Please let me know when you hear from them.

yetanotherusern4me, 2024-06-20:
I am having this same issue with raspberry pi 4 image.

abheek648, 2024-07-07:
I have the same problem with kali

cxnturi0n, 2024-07-12:
This is still an Issue. The image seems to be corrupted, from either rpi installer and from https://www.kali.org/get-kali/#kali-arm

sam-david, 2024-07-12:
@tdewey-rpi any updates from the Kali team? 

For anyone else, I got around this by flashing the ARM images directly from Kali at https://www.kali.org/get-kali/#kali-arm

abheek648, 2024-07-13:
@sam-david is the image 64bit or 32bit?

andychess, 2024-07-14:
Hi,
Same problem here.  Both 32 and 64 bit images affected.  It's not only a problem on the Rpi imager, downloading directly from the Kali website also produces the error.

sam-david, 2024-07-15:
> @sam-david is the image 64bit or 32bit?

I understand a lot of you are having issues even with directly from Kali website. However, this worked for me on my Raspberry Pi 5s using the 2024.2 arm64 download.

andychess, 2024-07-15:
I just tried downloading the Pi5 version and can confirm that it is working fine.  The problem seems to be with the 32 and 64 bit  Pi 2, 3, 4, 400 images.

I my case I have tried on Mac / Win using Balena Etcher and the Pi Imager. Using tar, gunzip etc all produce an error.

abheek648, 2024-07-15:
I hope this link helps cuz it helped me  https://www.youtube.com/watch?v=ZzCdfey8abU

sam-david, 2024-07-15:
> I just tried downloading the Pi5 version and can confirm that it is working fine. The problem seems to be with the 32 and 64 bit Pi 2, 3, 4, 400 images.
> 
> I my case I have tried on Mac / Win using Balena Etcher and the Pi Imager. Using tar, gunzip etc all produce an error.

Yes, @andychess makes a good point I forgot to mention. I used Balena Etcher instead of the Pi Imager to flash for the direct kali downloads

NenadX, 2024-07-17:
I also have a problem using dd on Kali Vbox, as well as on the net installation.
[image]