raspberrypi/rpi-imager #886: [BUG]: Can't See Options on Imager

Daiaro, 2024-06-27:
### What happened?

I am new to raspberry pi so I may be missing something simple. All I did was download the setup exe from https://www.raspberrypi.com/software/. Installed it and I can not see any options from the first menu that pops up when you run the software. I tried just clicking through options by looking through online guides but it does not do anything. Thanks for any help and here is some system info:

Processor	Intel(R) Core(TM) i5-6500T CPU @ 2.50GHz   2.50 GHz
Installed RAM	16.0 GB (15.9 GB usable)
System type	64-bit operating system, x64-based processor
Edition		Windows 10 Enterprise
Version		22H2
Display 	        Samsung TV model#: UN40EH6000F

[image]
](url)

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

22H2

### Selected OS

Raspberry Pi OS

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-06-27:
Thanks for the report, @Daiaro.

This is a strange failure - it looks like Qt has decided to not render any text at all, which isn't what I'd expect. When you press the buttons , do the dialogs appear?

Daiaro, 2024-07-01:
Yes sometimes it does but other times it does not respond. For instance, I tried pressing the third button which should be to select the type of storage and it showed a blank page with nothing on it. Another time when I reinstalled it the button does nothing. Seems to be random.

lurch, 2024-07-01:
Can you check that the drivers for your graphics card are up to date please?

tdewey-rpi, 2024-07-01:
@Daiaro Inconsistent behaviour is even more unexpected!

Can you please post a screenshot from **dxdiag**? (https://www.windowscentral.com/how-check-your-computer-full-specifications-windows-10#section-how-to-find-computer-specs-from-directx-diagnostic-tool)

Daiaro, 2024-07-03:
> Can you check that the drivers for your graphics card are up to date please?

Yup that was it whoops. To be fair I can't install updates at work IT had to do it. I just assumed it was already up-to-date. Thanks for the replies.

tdewey-rpi, 2024-07-31:
Closing as 'Won't Fix' - because it was the OS!

closed 2024-07-31 by tdewey-rpi