raspberrypi/rpi-imager #887: [BUG]: keyboard configuration is not applied

SushiFanta, 2024-06-27:
### What happened?

see this [forum post](https://forums.raspberrypi.com/viewtopic.php?p=2232354#p2232354).

When keyboard setting is configured in the imager, it changes the appearance of the setting in raspi-config but doesn't actually apply the setting to change the keyboard behavior. As a result, any preconfigured image will type in br*tish characters regardless of the setting until manually changed.

I am investigating the cause and trying to find a way to either fix the imager or create a script to fix it after boot.

### Version

1.8.5 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

Ubuntu 24.04 LTS

### Selected OS

Raspberry Pi OS Bookworm 64-bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-06-28:
Thanks for the report, @SushiFanta

This is a fairly concerning failure - can you confirm it's fundamentally different from the request in #852?

SushiFanta, 2024-06-28:
It's different than #852, it was not affected when I was changing locales around. After digging further, it seems to be like the issue from #693. The only difference is that my raspi-config at least displays the correct layout, although it still isn't applied. They suggested waiting for an update to netinstall imager that should support this customization for bookworm, but it seems like this problem made it past the update from a few days ago.