raspberrypi/rpi-imager #888: [BUG]: Unable to instantiate a Home Assistant for Raspberry Pi 5 on Windows

qrnch-jan, 2024-06-30:
### What happened?

Using the Windows version to instantiate a Raspberry Pi 5 micro-SD card yielded an unbootable SD card. We tried a different micro-SD card but got the same result.  We retried three or four times with the same results.  We had a Raspberry Pi 3 laying around, so we tried to instantiate Home Assistant for that instead, which worked fine (it booted).

Last week I successfully built a Raspberry Pi 5 image of Home Assistant from a mac, so I knew it should work.

We switched to a mac, and that built the RaspberryPi 5 variant of Home Assistant that would boot correctly.

There seems to be a _Windows_-specific bug that causes instantiated _Home Assistant_ images _for Raspberry Pi 5_ to not be bootable.

("Not bootable" means that the first stage bootloader complains that it can't detect a bootable image)

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

10

### Selected OS

Home Assistant

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-07-01:
Sounds like this might be the same thing as https://github.com/raspberrypi/rpi-eeprom/issues/585 which says "Turned out the Imager itself isn't to blame, but the Windows kernel (presumably on Windows 10+, Windows 7 don't show that behavior), which alters the partition table **even when a drive is just plugged into the computer**."

qrnch-jan, 2024-07-01:
Yeah, that definitely looks like it could be the same issue.

What an odd thing for the kernel to do -- would be interesting to learn what the origin of that is.

tdewey-rpi, 2024-08-28:
Closing as 'Won't Fix'.

Unfortunately, I don't have the power to convince Redmond to change Windows system behaviour - though if anyone is aware of a workaround or change that I'm not aware of I'm prepared to review a PR.

closed 2024-08-28 by tdewey-rpi