raspberrypi/rpi-imager #893: [BUG]: Settings window offers none at all

SvenMichaelKlose, 2024-07-14:
### What happened?

After selecting the Raspi-3A+ & Pi OS 64-bit, the settings window offers no choices.

### Version

1.8.5 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

Fedora 40

### Selected OS

Pi OS Bookworm 64-bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-07-15:
Thanks for the report, @SvenMichaelKlose 

We don't see this issue on any of the reference platforms or OSes. In your report, you stated:

> I've built the latest from this repository.

Is this report against v1.8.5, or HEAD?

SvenMichaelKlose, 2024-07-15:
I apologize for the confusion. It's HEAD.

SvenMichaelKlose, 2024-07-19:
Fixed as of c1c8c5a7dd1c5b433a9cb33df9b8989c55ccc9a8.

You're awesome!

closed 2024-07-19 by SvenMichaelKlose