raspberrypi/rpi-imager #894: [BUG]:

FiSh3rWoLf, 2024-07-16:
### What happened?

The network imager doesn't read IMAGER_REPO_URL in bootloader EEPROM.
Imager look for [rmem0](https://github.com/raspberrypi/rpi-imager/blob/686ad14308639b0589a2971903066f56271033e7/src/imagewriter.cpp#L90C25-L90C55) but has rmem4 instead.

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

23H2

### Selected OS

Network Imager (HTTP Boot)

### Which Raspberry Pi Device are you using?

Raspberry Pi 5

### What kind of storage device are you using?

NVMe SSD via Third-party PCIe attachment

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
Welcome to Buildroot
buildroot login: root
Password:
login[130]: root login on 'console'
# ls /sys/bus/nvmem/devices/
nvmem_cust0  nvmem_mac0   nvmem_otp0   nvmem_priv0  rmem4
```

tdewey-rpi, 2024-07-17:
Thanks for the report, @FiSh3rWoLf, and for the pre-submission sleuthing.

Curious we've not seen this one before. I'll need to check if this is specific to Raspberry Pi 5, or if it also affects the Raspberry Pi 4 family.

timg236, 2024-07-17:
Applications need to use the blconfig alias and not hardcoded the rmem node
https://github.com/raspberrypi/rpi-eeprom/blob/master/rpi-eeprom-update#L117

thePhil, 2024-10-05:
I can confirm that I'm hitting the same issue.

1. Original RPI Imager downloaded from the official location and served locally in my network
2. IMAGER_REPO_URL configured and no GET request can be recorded for the specified file

Would be glad to hear about a projected timeline for a resolution.
As a workaround, I will experiment, patching the URL I need into rpi-imager and serve that to my Raspberry Pi 5.

Please see the following context:
`boot.conf` file of the bootloader:
```
[all]
ENABLE_SELF_UPDATE=1
BOOT_UART=1
POWER_OFF_ON_HALT=0
BOOT_ORDER=0xf271

HTTP_HOST=192.168.178.19
HTTP_PORT=18080
HTTP_PATH=/
IMAGER_REPO_URL=http://192.168.178.19:18080/os_list_imagingutility_v3.json
```

Boot UART Trace:
```
91.68 HTTP: GET request for http://192.168.178.19:18080/boot.img
191.72 HTTP: got 98% at 183 kB/s
192.43 HTTP: got 98% at 183 kB/s
193.71 HTTP: got 98% at 182 kB/s
194.35 HTTP: got 99% at 183 kB/s
196.27 HTTP: got 99% at 181 kB/s
196.72 HTTP: got 100% at 181 kB/s
196.73 Verifying
200.61 RSA verify
200.73 rsa-verify pass (0x0)
200.17 MBR: 0x00000000,       0 type: 0x00
200.18 MBR: 0x00000000,       0 type: 0x00
200.22 MBR: 0x00000000,       0 type: 0x00
200.26 MBR: 0x00000000,       0 type: 0x00
200.29 Trying partition: 0
200.32 type: 16 lba: 0 'mkfs.fat' '  V       ^ ' clusters 15851 (4)
200.38 rsc 4 fat-sectors 64 root dir cluster 1 sectors 16 entries 256
200.44 FAT16 clusters 15851
200.48 Read config.txt bytes      207 hnd 0xb70
200.51 [ramdisk] pieeprom.upd not found
200.55 usb_max_current_enable default 0 max-current 5000
200.61 Read bcm2712-rpi-5-b.dtb bytes    80218 hnd 0xace
200.65 dt-match: compatible: raspberrypi,5-model-b match: brcm,bcm2712
200.71 dt-match: compatible: brcm,bcm2712 match: brcm,bcm2712

NOTICE:  BL31: v2.6(release):v2.6-239-g2a9ede0bd
NOTICE:  BL31: Built : 14:26:57, Jun 22 2023
Starting rpi-imager

Welcome to Buildroot
buildroot login: 
```

Respective network flow observed via Wireshark:
```
No.	Time	Source	Destination	Protocol	Length	Info
630181	8737.626322	192.168.178.183	192.168.178.19	HTTP	194	GET /boot.sig HTTP/1.1 
630184	8737.628228	192.168.178.19	192.168.178.183	HTTP	144	HTTP/1.1 200 OK  (application/pgp-signature)
630209	8738.827228	192.168.178.183	192.168.178.19	HTTP	194	GET /boot.img HTTP/1.1 
631234	8739.827270	192.168.178.183	192.168.178.19	HTTP	254	GET /boot.img HTTP/1.1 
634098	8747.327258	192.168.178.183	192.168.178.19	HTTP	255	GET /boot.img HTTP/1.1 
637960	8765.827341	192.168.178.183	192.168.178.19	HTTP	255	GET /boot.img HTTP/1.1 
642869	8772.827284	192.168.178.183	192.168.178.19	HTTP	255	GET /boot.img HTTP/1.1 
646519	8788.327386	192.168.178.183	192.168.178.19	HTTP	255	GET /boot.img HTTP/1.1 
651163	8805.881819	192.168.178.183	192.168.178.19	HTTP	254	GET /boot.img HTTP/1.1 
653685	8806.827512	192.168.178.183	192.168.178.19	HTTP	254	GET /boot.img HTTP/1.1 
655597	8824.565430	192.168.178.1	255.255.255.255	DHCP	342	DHCP ACK      - Transaction ID 0x5c59d54a
664389	8847.327670	192.168.178.183	192.168.178.19	HTTP	255	GET /boot.img HTTP/1.1 
668009	8877.367097	192.168.178.183	192.168.178.19	HTTP	254	GET /boot.img HTTP/1.1 
668604	8878.327985	192.168.178.183	192.168.178.19	HTTP	255	GET /boot.img HTTP/1.1 
674494	8888.327771	192.168.178.183	192.168.178.19	HTTP	254	GET /boot.img HTTP/1.1 
677947	8894.327689	192.168.178.183	192.168.178.19	HTTP	255	GET /boot.img HTTP/1.1 
686558	8908.327763	192.168.178.183	192.168.178.19	HTTP	254	GET /boot.img HTTP/1.1 
687499	8910.233172	192.168.178.19	192.168.178.183	HTTP	619	HTTP/1.1 206 Partial Content  (application/octet-stream)
```

lurch, 2025-01-06:
@tdewey-rpi Perhaps this issue should be left open until an updated version of the embedded imager has actually been published? :thinking:  (Which I think means that you'll also need to update the Qt version used in the embedded imager? :sweat:  It didn't build for me when I tried building the embedded imager shortly before Christmas.)

bmidy, 2026-03-01:
no news ?
is there any date for the new officially fixed version release

PR #976 (2025-01-06): embedded: Use blconfig alias to find bootloader config
Use the blconfig alias to find the correct nvmem device to read the bootloader configuration, the practical effect being that IMAGER_REPO_URL should work on Raspberry Pi 5.

Fixes #894