raspberrypi/rpi-imager #897: [BUG]: Imager v1.8.5 doesn't run properly when using bookworm on Raspberry Pi 2 Model B Rev 1.1

drghughes, 2024-07-23:
### What happened?

The Raspberry Pi Imager v1.8.5 doesn't run properly when using bookworm on my Raspberry Pi 2 Model B Rev 1.1. The Imager window doesn't display anything - the window is just black/white and the pink/red colour used in the Imager (see screenshot).  Everything is up to date.

[image]

Just in case it is important, I am running the Pi headless and connecting using RealVNC. Apart from the headless/RealVNC usage, I don't believe that I have made any changes to settings. The installation is relatively recent, and was done using the 15 March 2024 Raspberry Pi OS with desktop image.

Things are a little better if I expand the Imager window to full screen, but the window is still distorted.  I haven't bothered to see if I can make it work like this.

[image]

### Version

1.8.5 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

bookworm

### Selected OS

Didn't get that far

### Which Raspberry Pi Device are you using?

Raspberry Pi A, B, A+, B+, and Compute Module 1

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-07-23:
FYI @tdewey-rpi  I can confirm this behaviour.

On a Pi 2B running Bookworm and X11 with an HDMI monitor plugged in, RPi Imager displays fine on the monitor. If I then connect to the Pi 2B with VNC, then RPi Imager still displays fine. But if I unplug the HDMI monitor, reboot the Pi, connect over VNC and run RPi Imager again, then I see a similar "corrupted" display as shown above, so it looks like some kind of incompatibility between the QT rendering in RPi Imager and the RealVNC server? (but _only_ when RealVNC is in headless mode) :thinking:

tdewey-rpi, 2024-07-26:
@drghughes Ah, you must have missed the announcement:

https://www.raspberrypi.com/news/raspberry-pi-connect-remote-shell-access-and-support-for-older-devices/

drghughes, 2024-07-26:
@tdewey-rpi, I've installed rpi-connect and labwc and restarted, but the Connect icon hasn't appeared.

Possibly this is because I'm using RealVNC??  And unfortunately I can't try connecting an HDMI monitor as I don't own one.

So am I stuck, or can I get around this using the command line?  I don't want to muck up this installation, so don't want to risk anything.

tdewey-rpi, 2024-07-26:
> @tdewey-rpi, I've installed rpi-connect and labwc and restarted, but the Connect icon hasn't appeared.
>
> Possibly this is because I'm using RealVNC?? And unfortunately I can't try connecting an HDMI [...]

I'd suggest you hold off, then, and I'll ask @lurch to take a look at the Connect use case.

For reference: installation of labwc would not be sufficient, as you'd also need to use `raspi-config` to enforce use of labwc. Doing so would effectively break RealVNC - as it would remove the X server from running - and you'd be left with a Raspberry Pi Connect exclusive environment. You would, in theory, be able to switch _back_ to X11 using `raspi-config` via Raspberry Pi Connect Remote Shell, but I've not tried this recently.

TenoTrash, 2024-10-25:
Hello there, I have the very same problem, Imager works very bad with VNC, and I was not able to create a fully working image using only the imager from the command line

lurch, 2024-10-27:
@TenoTrash Rather then trying to run Imager on the (headless?) Raspberry Pi and view its output via VNC, are you able to run Imager on the same computer that you're running your VNC client on?

lurch, 2024-10-27:
> I'd suggest you hold off, then, and I'll ask @lurch to take a look at the Connect use case.

Sorry @tdewey-rpi , I completely forgot about this! I've just checked, and on a headless Raspberry Pi 2B that's running the labwc compositor and Connect screen-sharing, Imager displays and runs totally fine.

TenoTrash, 2024-10-28:
> @TenoTrash Rather then trying to run Imager on the (headless?) Raspberry Pi and view its output via VNC, are you able to run Imager on the same computer that you're running your VNC client on?

We use mostly Android tablets and cellphones to log into Raspberry's DE. We solve keeping a pre-configured image in a pendrive and simple dd the image into fresh SD card in case of need, but it will be nice to be able to run imager

TenoTrash, 2024-10-29:
> @TenoTrash Rather then trying to run Imager on the (headless?) Raspberry Pi and view its output via VNC, are you able to run Imager on the same computer that you're running your VNC client on?

I'm happy to announce that with the last update, and being able to migrate from x11 to wayland in our fleet of Raspberry Pi 3+, Imager works as expected.
This is a very good news to our people, who has very limited access to technology and resources, plus moving computer related parts to remote areas.
Thank you all for the work and support, we really apreciate it.

tdewey-rpi, 2024-10-29:
@TenoTrash That's excellent news - thank you for the report. I'm not going to close this issue until the original reporter has had opportunity to try out the new Raspberry Pi OS and hopefully also report success.

Holding open for the next 2 weeks.

drghughes, 2024-11-05:
Finally got to this.  I updated everything according to the instructions in [https://www.raspberrypi.com/news/a-new-release-of-raspberry-pi-os/](https://www.raspberrypi.com/news/a-new-release-of-raspberry-pi-os/) (though it seemed that a normal upgrade was all that was needed, as nothing extra was added when I did the full-upgrade). and then switched to labwc when I got the prompt.

When I restarted the Pi, all I got was a tiny black window in VNC.  I used PuTTY to connect and eventually sorted everything out.  Wayland wasn't much better than labwc (grey and bigger, but otherwise nothing) and I had to go back to X to be able to get things working.

Imager still doesn't work under X.

Maybe everything would work if I did a fresh install, or if I switched to Connect, but I don't have time to play with this at the moment, so it's not something that I will be trying in the next few weeks (probably longer).