raspberrypi/rpi-imager #898: [BUG]: When bhyve virtualizes Linux Ubuntu,the Raspberry PI Imager does not recognizes any USB stick / sd card passed thru bhyve

Marietto2008, 2024-07-28:
### What happened?

Hello.

I'm running Ubuntu 24.04 virtualized with bhyve (BSD hypervisor) using these parameters :
```

/usr/sbin/./bhyve-lin -S -c sockets=2,cores=2,threads=2 -m 8G -w -H -A \
-s 0,hostbridge \

-s 1,ahci-hd,/mnt/$vmdisk4'p2'/bhyve/img/Linux/Ubuntu-2404-Android.img,bootindex=1 \

-s 2,ahci-hd,/dev/$vmdisk15 \

-s 8:0,passthru,2/0/0 \
-s 8:1,passthru,2/0/1 \
-s 8:2,passthru,2/0/2 \
-s 8:3,passthru,2/0/3 \
-s 11,hda,play=/dev/dsp,rec=/dev/dsp \
-s 13,virtio-net,tap9 \
-s 14,virtio-9p,sharename=/ \
-s 29,fbuf,tcp=0.0.0.0:5909,w=1600,h=950,wait \
-s 30,xhci,tablet \
-s 31,lpc \
-l bootrom,/usr/local/share/uefi-firmware/BHYVE_UEFI_CODE.fd \
vm0:9 < /dev/null & sleep 2 && vncviewer 0:9
```

Where /dev/$vmdisk15 is :
```

Disk /dev/sdb: 953,68 GiB, 1024005767168 bytes, 2000011264 sectors
Disk model: BHYVE SATA DISK 
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: E0E79E7B-DC41-4DD7-9B8C-62B075FAF126

Dispositivo   Start       Fine    Settori   Size Tipo
/dev/sdb1      2048      34815      32768    16M ChromeOS kernel
/dev/sdb2     34816      67583      32768    16M ChromeOS kernel
/dev/sdb3     67584    1116159    1048576   512M Linux filesystem
/dev/sdb4   1116160 2000009215 1998893056 953,1G Linux filesystem
```

My intention was to write the Lineage OS to the sd card using the Raspberry PI Imager,without to rebooting and without using Linux natively. Well,it is not able to detect my sd card (that I have placed inside my SD card TO USB adapter).

The images below shows that the sd card passed through bhyve is not recognized by the RaspBerry Pi Imager,that freezes.

What can I do to unfreezes it ? Let us know,thanks.

[image]

### Version

1.8.5 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

24.04

### Selected OS

Ubuntu 24.04

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in a USB reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
There isn't.
```

Marietto2008, 2024-07-28:
Opened a bug ticket on the FreeBSD bug tracker :

https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=280481

lurch, 2024-07-29:
I see that you have:
```
Disk model: BHYVE SATA DISK 
```

IIRC the https://github.com/balena-io-modules/drivelist module used by RPi Imager assumes that SATA drives are "internal", and to avoid the user accidentally overwriting their main HDD, it only lists drives that are identified as "removable". So maybe you need to get your "passed thru" drive to appear as a USB drive rather than a SATA drive?

Marietto2008, 2024-07-29:
Bhyve can't pass thru usb drives. Its a not implemented function. I think that a patch that convert the disk from SATA to USB is needed.

tdewey-rpi, 2024-07-30:
Thanks for the report, @Marietto2008 

Unfortunately, there's no quick path to resolving this issue from our side.

In the longer term, we might look at adding a mechanism to stop excluding devices on the storage selection screen - but as this presents a hazard to new users and affects comparatively few users it's much lower on my priority list than the upcoming v1.9.0 targets.

Marietto2008, 2024-07-30:
Sometime ago I have reported the same problem,but for Rufus. Even Rufus didn't recognize the bhyve disks. The problem has been fixed by this commit :

https://github.com/pbatard/rufus/pull/2504

who knows,maybe it can be useful for you as idea to implement.

tdewey-rpi, 2024-07-31:
Thanks for the pointer, @Marietto2008 

Unfortunately our linux target storage device filter works in a very different way - which is why I said there was no short path to resolution.

One thing that might help identify a resolution is to attach the output of the following command inside your VM:

`lsblk --bytes --json --paths --output-all --exclude 7`

tdewey-rpi, 2024-08-28:
Closing as stale.

Resolving this issue will require a degree of interactive debugging between myself and a FreeBSD user - and may only result in _one_ emulated environment having the right behaviour, due to the nature of the DriveList code.

As I've had no feedback in a month, and I do not otherwise have access to FreeBSD, I'm making this as 'Stale' to focus on other features and the upcoming v1.9.0.

closed 2024-08-28 by tdewey-rpi

Marietto2008, 2024-08-28:
This is the right FreeBSD person who can help,but I don't know how to contact him :

https://github.com/markpeek