raspberrypi/rpi-imager #900: [BUG]: Popup "OS customisation menu"  is not displayed

alpineman99, 2024-07-31:
### What happened?

My win 11 Laptop is connected to a monitor. I set the monitor as the primary display in the windows settings > system > display
I opened the raspberry pi imager v1.8.5 and moved the window to the secondary laptop screen. Everything worked until I wanted to edit settings in the "use OS customisation" popup. I clicked on the "edit settings" button, but the OS customisation menu did not popup. By chance I moved the rpi imager window to the primary monitor. There everything worked as it should.

Issue:
- The OS customisation menu doesn’t popup, if rpi imager window is placed on secondary display.

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

23H2

### Selected OS

Raspberry Pi OS (32-bit)

### Which Raspberry Pi Device are you using?

Raspberry Pi 3B, 3A+, 3B+, and Compute Module 3, 3+

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

tdewey-rpi, 2024-08-01:
Thanks for the report, @alpineman99 

This is an issue of the sort I'd expect with v1.8.5, which uses Qt5 - we've seen a range of multi-display problems, particularly when different displays use different pixel scaling factors.

v1.9.0 will use Qt6, which I'm expecting to have much better behaviour.

tdewey-rpi, 2024-09-11:
@alpineman99 Please try v1.9.0, and let me know if you're still having the problem

alpineman99, 2024-09-26:
@tdewey-rpi v1.9.0 fixed the issue. Thank you for solving it.

tdewey-rpi, 2024-09-26:
Thanks for letting me know, @alpineman99.

Glad to see another UI problem gone!

Closing as 'Fixed'.

closed 2024-09-26 by tdewey-rpi