raspberrypi/rpi-imager #901: [BUG]: PI 4 doesn't connect to wifi

thewh1teagle, 2024-08-04:
### What happened?

It may be an issue with Raspbian itself. 
I just flashed raspbian lite 64 to RPI4, pre configured the wifi and boot.
It doesn't connect to the wifi.
I verified that the name & password are correct.
I'm not sure how to diagnose it to know better if it's an issue with rpi imager.

I can access the shell with keyboard & hdmi.

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Windows

### Selected OS

Bookworm 64

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

USB SSD

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

archfrog, 2024-08-08:
I'm having the same problem, with Bookworm 64, but I'm am a Debian noob.  So I switched to Ubuntu Server 24.04 and it works with the exact same OS customizations as was used for Debian.

This just to tell the developers that the problem probably lies with the Raspbian OS support as Ubuntu works as it should.  If I try to use `raspi-config` to specify the already correct wireless SSID and key, I get some obscure error (on Bookworm).

archfrog, 2024-08-08:
Just stumbled across this passage while checking out some RPi kiosk software:

["RPi OS bookworm dropped wpa_supplicant.conf support for configuring WiFi and moved to Network Manager."](https://github.com/guysoft/FullPageOS/issues/585#issuecomment-2275423610)

Guess that explains why configuration of WIFI in RPI Imager no longer works on Bookworm.

Vizzyy, 2024-08-10:
I'm pretty sure this is related... but I just wasted like 5 hours troubleshooting (countless reflashes of bookworm, bullseye, separate machines, sdcards, etc) being unable to SSH into a fresh RPI OS image (via rpi-imager, setting a default user, enabling ssh). 

Headless SSH works the INITIAL boot, but then any following boot I am not able to SSH until I first sign into the pi locally (via monitor and keyboard), and then I am able to headless ssh from another machine. Clearly there is something going on with Network Manager that does not initialize the connectivity unless the user is already logged in.

I was able to fix this by leveraging the `nmtui` GUI tool to edit my `preconfigured` connection setup through rpi-imager. There is a checkmark for `Available to all users` that needs to be enabled. For me this was already enabled by rpi-imager, and so I had to disable it and then reenable it and now I am finally able to ssh headlessly every boot. 

My money is on rpi-imager doing something wonky with this `preconfigured` connection setup during the flashing.

XECDesign, 2024-08-12:
> Guess that explains why configuration of WIFI in RPI Imager no longer works on Bookworm.

Nope, it should still work.

Output of `sudo journalctl` from the non-working system would help figure out what's going on.

> For me this was already enabled by rpi-imager, and so I had to disable it and then reenable it and now I am finally able to ssh headlessly every boot.

Has the content of  `/etc/NetworkManager/system-connections/preconfigured.nmconnection` changed after you've made that change?

remohoeppli, 2024-08-13:
I see the same issue.

Here's an output of the `journalctl`.
[image]

My `/etc/NetworkManager/system-connections/preconfigured.nmconnection` does contain a configuration.

I tried the latest Debian Bookworm 64-Bit Lite and Full version. I hope this helps.

XECDesign, 2024-08-13:
Thank you. Another potentially useful data point. Was it working in the 2023-10-10 image and then broken in the 2023-12-05 image?

GierAffe, 2024-08-15:
After writing the image to a micro sd card (or usb etc), a firstrun.sh file should have been created on the boot partition. There is a line:
```
if [ -f /usr/lib/raspberrypi-sys-mods/imager_custom ]; then
   /usr/lib/raspberrypi-sys-mods/imager_custom set_wlan '[my ssid]' '[my psk]' '[my country code]'
```
can you check if the psk is correct?
In a linux shell, the command `wpa_passphrase [ssid] [passphrase]` will generate your psk. Do you get the same hash? If not, does it work if you replace the psk in the firstrun.hs file with the one you generated by hand?

You can also check if the file /etc/NetworkManager/system-connections/preconfigured.nmconnection has been generated (which it should).

ryepup, 2024-08-17:
just ran into this using `2024-07-04-raspios-bookworm-arm64-lite.img.xz`, `rpi-imager` v1.7.2, and a PI 5. I'm using Ubuntu 22.04 LTS, and have installed `rpi-imager` via `apt install`.

I did not have a `/etc/NetworkManager/system-connections/preconfigured.nmconnection` file, and used `raspi-config` to get connected.

I think my problem is that Ubuntu 22.04 LTS package repositories are delivering a really old version of rpi-imager. My `firstuse.sh` did not have anything about `raspberrypi-sys-mods`.

manjushsh, 2024-08-22:
+1 for rpi 5

tdewey-rpi, 2024-08-28:
> just ran into this using `2024-07-04-raspios-bookworm-arm64-lite.img.xz`, `rpi-imager` v1.7.2, and a PI 5. I'm using Ubuntu 22.04 LTS, and have installed `rpi-imager` via `apt install`.
>
> I did not [...]

As a general point, I strongly recommend that users do _not_ use `apt install` on platforms other than Raspberry Pi OS.

v1.7.2 was released before Raspberry Pi 5, and before the Bookworm release - so it's perfectly expected that it would fail to configure for new hardware and new software.

--

Are any of these failing PSKs using non-ASCII characters?

archfrog, 2024-08-29:
> As a general point, I strongly recommend that users do not use `apt install` on platforms other than Raspberry Pi OS.

You mean `apt install rpi-imager`, not `apt install` in general, right?

> Are any of these failing PSKs using non-ASCII characters?

Not here.

drewtoto, 2024-10-25:
Host OS Version
MacOS

Selected OS
Bookworm 64, Bookworm 32, Bookworm Lite

Which Raspberry Pi Device are you using?
Raspberry Pi 4B

What kind of storage device are you using?
Micro SD

OS Customisation
Yes, enabled wifi and ssh.

Comments
Also tried editing the _psk_ value in `firstrun.sh` and replacing the hashed password with plain text.

> Are any of these failing PSKs using non-ASCII characters?

Just ascii chars.

efkann, 2024-10-31:
I have a Raspberry Pi 3 Model B.

I installed Raspberry Pi OS, Bookworm64 with rpi-imager on my macOS using a Micro SD, setting up WiFi and everything.

Note: After the installation, I changed my locale from `tr` to `en-US`. I dont know if this is related to this issue though.

Everything was working perfect. After a power outage, the WiFi didn't want to reconnect, prompting me to enter my password again and again even though I was sure the password was correct.

After hours of troubleshotting I decided to remove the underscore in my SSID. I also removed the `*` and `#` characters in the password.

I reconnected to my new WiFi using `nmtui` and it worked, hope this helps.

octalparrot, 2024-11-19:
Rpi 3A+

Installed bookworm lite 32bit, windows rpi-imager, setting wifi, rsa keys.

Plugged SDcard to RPi, no wifi connection, plugged hdmi monitor and keyboard, shows ip addr 127.0.0.1

Managed to enter cmd "nmcli radio wifi on" - working fine even afer reboot

gasparramoa, 2024-11-19:
I also have this problem. Rpi 4B.

I lost around ~4 hours flashing and trying different setups, from setting wpa_supplicant.conf to setting the NetworkManager .nmconnection.

Bullseyes LITE 32-bit can set the wifi and ssh connection with the Imager.
Bookworm LITE 32-bit or 64-bit never connects to wifi. 
I am using Imager 1.8.5, which I downloaded for Windows. https://www.raspberrypi.com/software/

Is this problem already fixed for newer build versions (which I can build from the git repo)

Thanks

PaulCampbell, 2024-11-20:
I am seeing the same here 
Pi OS Lite (32 bit) Bookworm released 2024-11-13
on Pi Zero 2Ws

tdewey-rpi, 2024-11-20:
Could you please try against the Bookworm release from 19th November?

PaulCampbell, 2024-11-20:
@tdewey-rpi The latest release is working for me. Thanks!

`Pi OS Lite (32 bit) Bookworm 2024-11-19`

gasparramoa, 2024-11-20:
Same here! 
The 19th November release is also working for me! 

I am using `Imager 1.8.5` and `Raspberry Pi 4B`.
One note: I had to power cycle the device by unplugging and reconnecting the power cable. On the second boot, it connected to the Wi-Fi successfully!

Thank you @tdewey-rpi !!

tdewey-rpi, 2024-11-20:
While I'd love to accept the credit, true credit must go to @XECDesign, who has championed this (and additionally, drawn my attention more acutely). We'll be introducing steps to avoid this sort of issue going forward - with the hope you won't even notice that being introduced!

thewh1teagle, 2024-12-08:
> Could you please try against the Bookworm release from 19th November?

The latest release connected successfully to the WiFI. but now there's another issue that introduced: the SSH service is not enabled even though I enabled it in RPI imager.

lurch, 2025-01-29:
> I just realized that using that console command utility I was trying with ssid and password for wireless band my raspberry (4B in my case) does not detect or compatible for (?).

@fedebuyito If you're reporting a problem with the console command line utility not being able to connect to your WiFi, then that sounds like an OS-level problem, rather than a problem with Raspberry Pi Imager 😉 
Please open a new issue at https://github.com/raspberrypi/bookworm-feedback/issues explaining your problem in as much detail as possible.

CompThing, 2025-03-28:
I tried a few times to set up my Pi Zero 2 W using rpi-imager to set up WiFi.
This was unsuccessful until I discovered the the PSK in the image was not correct according to https://www.wireshark.org/tools/wpa-psk.html

Issue eventually tracked down to the calculation of the PSK from password and BSSID.
I have a dual band WiFi Access Point that supports roaming. the 2.4G and 5G WiFi have the same SSID, but a different BSSID.
The PSK was being calculated using the BSSID of the 5G Wifi (channel 100) rather than the 2.4G WiFi (channel 9).
I can't remember the exact steps I took to resolve the issue, and do not have a spare SD card to repeat.
I think I fixed things by calculating the correct PSK for the 2.4G BSSID and then editing the SD card from my host computer.
Hope that helps someone. I will try to find some time to come back with better info.

efkann, 2025-03-29:
> I have a Raspberry Pi 3 Model B.
>
> I installed Raspberry Pi OS, Bookworm64 with rpi-imager on my macOS using a Micro SD, setting up WiFi and everything.
>
> Note: After the installation, I changed my [...]

After some outages, my connection started dropping again. I spent hours debugging, replacing every component in my setup, and was about to lose my mind. Guess what? I switched to an HDMI display, and everything worked. Thinking the issue was my VGA to HDMI adapter, I replaced it with a new one—but it still wouldn’t connect. That’s when I became certain the adapter was the root cause.

Out of curiosity, I wrapped the adapter in aluminum foil and secured it with tape. To my surprise, both the old and new VGA to HDMI adapters worked flawlessly as long as they were covered. I suspect these cheap adapters interfere with WiFi signals, but I’m not entirely sure. Hopefully, my experience saves someone else from hours of frustration!

closed 2025-05-01 by tdewey-rpi

minisbett, 2025-06-08:
Cannot confirm I think? I tried both Bookworm 64-bit and Bullseye 32-bit, but configuring wifi does not allow me to SSH into the rpi. When I had plugged it into LAN and manually configured wifi it did work though.

Using a Raspberry Pi 4, and just downloaded rpi-imager.