raspberrypi/rpi-imager #905: [FEATURE]: Add a Are you sure you would like to write to?

kdsands88, 2024-08-15:
### Is your feature request related to a problem? Please describe.

Issue: Raspberry Pi Imager wrote an SD image to an external USB storage drive not the SD card reader.

  

### Describe the solution you would like to see implemented

Request: If I was given a question "Are you sure you would want to write to XXX with XXX amount of data?"  I would have seen I choose the wrong drive.  

### Describe alternatives you've considered

Also, Raspberry Pi Imager only shows the drive letter not the drive name.  If it show "you would want to write to Personal Data"  and not "E:\"  I would have know I was writing to the wrong drive!!!!

### Additional context

Raspberry Pi Imager currently makes it hard to indentify which drive I am writing to.  I only use this software once every 2 years.
[image]

### Version

1.8.5 (Default)

tdewey-rpi, 2024-08-15:
Thanks for the report, @kdsands88 

I can see some solid user experience improvements we could add as cues:
1) Take your suggestion to get the Volume Label, rather than just the mount point.
2) Perform a basic sorting, where we push SD* Card to the top of the list
3) Add an explicit confirmation for drives over 512GB.

I've just checked our DriveList dependency, and it looks like this functionality is not a straightforward addition - Drivelist is not obtaining the volume name today, so we'll need to teach it how to.

Feature request accepted.

lurch, 2024-08-15:
> I've just checked our DriveList dependency, and it looks like this functionality is not a straightforward addition - Drivelist is not obtaining the volume label* today, so we'll need to teach it how to.

https://github.com/balena-io-modules/drivelist/issues/123 and https://github.com/balena-io-modules/drivelist/issues/323 and https://github.com/balena-io-modules/drivelist/issues/402 :wink:

tdewey-rpi, 2024-08-15:
I'm not surprised that this hasn't been implemented - it's conceptually complicated.

At the media selection level, we're looking to select a *physical* device, but on on any *physical* device there may be multiple *volumes*. The *volumes* have labels.

By convention, Windows didn't expose partitioning on SD cards or USB drives, so you would generally expect a SD card or USB drive to have a single volume filling the entire device.

That's not the case after you've used Imager, though - we write a full disk image with at least two volumes.

In exposing this function in Imager, I'll need to be careful to capture not just the _first_ volume, but any volumes that might be affected.

Interestingly, on Linux we probably get the volume names already - as these are encoded in the mount points, which Drivelist _does_ get today. macOS behaves similarly - so it's just Windows, which uses the lossy Drive Letter assignment, that doesn't get the full details.

maxnet, 2024-08-15:
>I'm not surprised that this hasn't been implemented - it's conceptually complicated.
>At the media selection level, we're looking to select a physical device, but on on any physical device there may be multiple 
>volumes. The volumes have labels.

Drivelist already provides a mountpointLabels vector<string> ready for someone to toss all volumes you find on that disk in.

So don't see the complication.
Filling that on Windows just needs to be implemented.
GetVolumeInformationA() on all mount points will probably do what you want.