raspberrypi/rpi-imager #907: [BUG]: Storage Capacity is not large enough.  Needs to be at least 30.8 GB.  This on a target 30.8 GB USB drive.

roblatour, 2024-08-17:
### What happened?

A bug happened!

I tried to restore an image of a USB drive made via the linux DD command.  The source and the target USB drive are the same make and model.

Raspberry Pi Imager is reporting the target USB drive's capacity as 30.8 GB, but when I try to restore to that drive I get an error saying 'Storage Capacity is not large enough.  Needs to be at least 30.8 GB.'.

[image]
[image]

I tried imager on both a RPi 4B and Windows, same error message.

### Version

1.8.5 (Default)

### What host operating system were you using?

Other Linux environment

### Host OS Version

Raspberry OS

### Selected OS

Linx 6.6.31-rpt-rpi-v8 aarch64

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

lurch, 2024-08-18:
It might be that your source image is e.g. 30.802 GB but your target drive is 30.801 GB? So the source image **is** bigger than your target drive (as confirmed by Etcher), but the GUI of Raspberry Pi Imager is rounding both sizes to "30.8 GB".

> It would be nice if rather than the RPi Imager saying there is an error and not continuing, that it would say that the target is too small to copy the image onto, but then offer an option to auto resize the final partition (as the RPi copy utility does without asking) such that the image process can proceed.

The RPi copy utility only runs under Raspberry Pi OS, and so has access to all the usual Linux filesystem tools. But RPi Imager also runs under Windows and MacOS, where it's not possible to resize ext4 filesystems.

roblatour, 2024-08-18:
@lurch 

> "The RPi copy utility only runs under Raspberry Pi OS, and so has access to all the usual Linux filesystem tools. But RPi Imager also runs under Windows and MacOS, where it's not possible to resize ext4 filesystems."

fair enough, but for those running it on Raspberry Pi OS, and where the source is ext4, why not make this an option?

lurch, 2024-08-19:
I've not worked on Raspberry Pi Imager myself, but I believe a lot of effort has gone into making it function as similarly as possible across Linux, Windows and MacOS.

tdewey-rpi, 2024-08-23:
Thanks for the report, @roblatour 

I believe @lurch has already covered what has likely gone wrong, but I can comment on why partition resizing is not an option.

@lurch is again correct, in that `rpi-imager` is designed to run across a range of operating systems - some of which will not have the functionality to perform offline resizes of ext4 partitions, or easily work with full disk images. 

Generally, it is best practice to ship an OS image that is the minimum viable size, and then perform an online resize - which is what Raspberry Pi OS will do.

roblatour, 2024-08-23:
@tdewey-rpi Thank you.  Yes, I was later able to confirm one of the 32 GB USB drives had ~25 MB more capacity than the other.

Also, while I understand Raspberry Pi imager is designed to run across a range of operating systems - some of which will not have the functionality to perform offline resizes of ext4 partitions, or easily work with full disk images, I believe that it is most likely that Raspberry Pi Imager has its greatest user base in the Raspberry Pi / Linux world and thus this feature would be of benefit to most of its user base.

It is certainly not unheard of for software to offer additional options and features where the OS / hardware allow for it.  Myself I have been a developer for several decades and have made software available that does this exactly - i.e. test for the availability of a specialized feature and if available provide for it.   In fact, as I think on the matter, Imager does this already when it filters what OSs are presented based the device the user selects for it to flash.

Regardless, as a bug, this request may imo be closed - but if it can be carried forward as a feature request that would be great.  

With thanks.

roblatour, 2024-08-23:
Closing but opening a related feature request here: https://github.com/raspberrypi/rpi-imager/issues/910

closed 2024-08-23 by roblatour