raspberrypi/rpi-imager #908: [BUG]: Custom OS not booting

Quaso17, 2024-08-21:
### What happened?

I am currently trying to use my custom OS I used half a year ago already and boot up a new Raspberry Pi 4 Model V Rev 1.5. For some reason I end up in the initramfs. I tried another SD-Card, another Raspberry Pi, another OS and came to the conclusion that my SD-Card is fine and so is my Raspberry Pi. The other OS works perfectly fine but I dont think there is an issue with my custom OS as it already worked prior and hasnt been touched since. Ive tried resolving errors within initramfs but with no success. At this point im out of ideas.

### Version

1.8.5 (Default)

### What host operating system were you using?

Windows

### Host OS Version

22H2

### Selected OS

Custom

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
I got this when I used a command (I unfortunately forgot) in initramfs:

ext2fs_open2: Bad magic number in super-block
fsck.ext4: Superblock invalid, trying backup blocks...
fsck.ext4: Bad magic number in super-block while trying to open /dev/mmcblk0p2
The superblock could not be read or does not describe a valid ext2/ext3/ext4
filesystem. If the device is valid and it really contains an ext2/ext3/ext4
filesystem. (and not swap or ufs or something else), then the superblock:
is corrupt, and you might try running e2fsck with an alternate superblock:
e2fsck -b 8193 <device>
or
e2fsck -b 32768 <device>
fsck exited with status code 8
done.
Warning: File system check failed but did not detect errors
mount: mounting /dev/mmcblk0p2 on /root failed: Invalid argument
Failed to mount /dev/mmcblk0p2 as root file system.
```

timg236, 2024-08-21:
You've not provided any details of your custom OS. Given that all the other operating systems flash and boot successfully you would need to explain what's different in your custom OS.

Quaso17, 2024-08-21:
> You've not provided any details of your custom OS. Given that all the other operating systems flash and boot successfully you would need to explain what's different in your custom OS.

I didnt create the OS and only flased it onto the SD-Cards before. Its based of Debian Bookworm 64-bit from March this year.
Its merely pre-configurations, pre-installed software and thats about it.

tdewey-rpi, 2024-08-23:
Thanks for the report, @Quaso17.

Unfortunately, I cannot see how this can be an issue with Imager. `rpi-imager` does not offer customisation of custom OSs, and as such it would have performed only a very straightforward block copy.

From your log, it looks like the custom OS image file contains an invalid ext4 partition, and the log is telling you how to fix it:

```ext2fs_open2: Bad magic number in super-block
fsck.ext4: Superblock invalid, trying backup blocks...
fsck.ext4: Bad magic number in super-block while trying to open /dev/mmcblk0p2```

Have you confirmed the source file has not been corrupted, or modified in a manner you didn't expect?

Quaso17, 2024-08-26:
The custom image is based on Debian Bookworm 64-bit and only things that were done are: Enabled Kioskmode, made pre-configurations (hostname, ip etc…).  And 5 months ago when I tried writing the image on my SD-Card it worked perfectly fine. I tried different Pi’s, different SD-Cards and when I tried other OS they worked. The custom image has never been changed either.

Quaso17, 2024-08-26:
> Thanks for the report, @Quaso17.
>
> Unfortunately, I cannot see how this can be an issue with Imager. `rpi-imager` does not offer customisation of custom OSs, and as such it would have performed only [...]

The custom image is based on Debian Bookworm 64-bit and only things that were done are: Enabled Kioskmode, made pre-configurations (hostname, ip etc…).  And 5 months ago when I tried writing the image on my SD-Card it worked perfectly fine. I tried different Pi’s, different SD-Cards and when I tried other OS they worked. The custom image has never been changed either.

tdewey-rpi, 2024-08-30:
> I tried different Pi’s, different SD-Cards and when I tried other OS they worked. The custom image has never been changed either

I'd suggest that must be a paradox. Have you tried other OS writing tools, eg, Balena Etcher?

Quaso17, 2024-09-02:
> > I tried different Pi’s, different SD-Cards and when I tried other OS they worked. The custom image has never been changed either
> 
> I'd suggest that must be a paradox. Have you tried other OS writing tools, eg, Balena Etcher?

I have also tried Win32 Disk Imager but the result was the same.

tdewey-rpi, 2024-09-02:
Thanks for the update, @Quaso17

If you're seeing similar behaviour with other tools, I'd suggest that your problem is likely with the image itself. Have you been able to perform a full file re-hash and compare against a known good value?

tdewey-rpi, 2024-09-18:
No response in > 2 weeks, and no indication this is a problem with `rpi-imager`. Closing as stale.