raspberrypi/rpi-imager #909: [BUG]: Imager wont use custon settings

davidbannon, 2024-08-22:
### What happened?

Using Debian Bookworm on my laptop, v1.8.5 of the Imager, targeting (initially) a Pi Zero and RasPi OS without desktop.

I have configured a custom config, hostname, a proper user name (mine), ssh, wifi credentials, locale etc. Ticked the boxes and saved. But nothing I can do will make the Imager use those options, I get the RasPi default every time.

I have tried different (RasPi) OSs, different target system settings, OS both with and without Desktop. In every case I end up with a User called pi, a hostname determined by the set target device. 

Interestingly, it appears it does take notice of my wifi config, as it does connect to Access Point but a login is not possible (using either my user name nor 'pi').

I have used the imager in the past, successfully, but possibly previous versions. I have remove the imager, blown away the config and reinstalled.

Any suggestions ?

Davo

### Version

1.8.5 (Default)

### What host operating system were you using?

Debian and derivatives (eg Ubuntu)

### Host OS Version

Bookworm

### Selected OS

RasPi OS, based on Bookworm 64bit/32bit with and without desktop

### Which Raspberry Pi Device are you using?

Raspberry Pi Zero 2 W

### What kind of storage device are you using?

microSD Card in an internal reader

### OS Customisation

- [X] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
Nothing that looks relevant when running imager from command line, with and without --debug.
```

davidbannon, 2024-08-23:
OK, some further information. After the imager announces its finished and the sdcard has been remove and then inserted into the raspi, I boot the raspi.  As noted above, ssh login is not happening.  After some time, I power it down as I cannot logon to gracefully take it down.  I then return the sdcard to my laptop and observe -

There is file called firstrun.sh present in the boot partition. It appears to be tasked with altering those default settings and, then, when finished, deleting itself and a reboot should occur.  Its still there, and the setting have not changed because, either, it has not run, the cmdline.txt file still has a reference to it -
 `... systemd.run=/boot/firstrun.sh systemd.run_success_action=reboot....`
or, perhaps it has run, changes were made but no `sync` to flush them to disk and no reboot ?

As I am running here headless, hard to tell. I'll see if I can free up some extra hardware to run this SDCard in a 'headed' system.

Davo

davidbannon, 2024-08-23:
OK, very sorry, this is a bogus bug report !  All due to me making two very silly mistakes. I thought I was using my Raspberry Pi Zero 2w, in fact it was the Raspi Zero One.  And the One does not work very well with a 64bit OS.  It works perfectly with a 32bit one !

Sorry for the unnecessary noise.

Davo

closed 2024-08-23 by davidbannon

session-exe, 2026-09-02:
I was making the same mistake hahah, thanks : )