raspberrypi/rpi-imager #910: [FEATURE]: When flashing a drive and the target is too small offer the option to adjust the final partition size to fit when possible

roblatour, 2024-08-23:
### Is your feature request related to a problem? Please describe.

Recently I tried to flash an image created via the linux dd command on one USB drive to another.   Although the two drives  were the same make, model and advertised capacity (32GB), it turns out the target USB drive had marginally less capacity (~25 MB).   After some research I found this issue is quite common.   In anycase, the Raspberry Imager reported the target drive was too small and would not flash the drive.

### Describe the solution you would like to see implemented

With Raspberry Pi Imager, when in a situation where an image is being flashed to a drive, yet the target is too small, the rather than the RPi Imager saying there is an error and not continuing, have it say that the target is too small to copy the image onto, but then offer an option to auto resize the final partition (as the RPi copy utility does without asking) such that the image process can proceed.  This to be offered if and when possible - I know it will not always be possible, however it should be possible with ext4 formatted partitions when the imager is running on Linux.  

### Describe alternatives you've considered

Flash to a larger drive, and then use Raspberry Pi SD Copier to copy the flashed larger drive to a smaller one.   Very time consumptive.

### Additional context

For more information please see this bug report: https://github.com/raspberrypi/rpi-imager/issues/907

### Version

1.8.5 (Default)

lurch, 2024-08-23:
> Recently I tried to flash an image created via the linux dd command on one USB drive to another. Although the two drives were the same make, model and advertised capacity (32GB), it turns out the [...]

In the meantime, you might find https://github.com/Drewsif/PiShrink interesting? (I've not used it myself, but I've seen other people mention it).

tdewey-rpi, 2024-08-30:
Feature request rejected.

There are two major themes to this rejection: The use case doesn't seem like a pattern I'd want to encourage, and I am not prepared to bifurcate the user experience depending on the host OS - recall that Windows users still make up the vast majority of Raspberry Pi Imager's user base, and they would have the largest implementation penalty for this feature.

On the use case: The main conceivable use case I can see to produce this issue is taking a full disk backup from one device and replicating it to others. This is extraordinarily risky - modern operating systems perform device individualisation as part of their set up, including storing passwords, unique IDs, encryption keys and other sensitive materials. You may have files inside that OS that you want to transfer - but doing so in this manner will also potentially leak those materials.

Instead, please consider using `pi-gen` or another similar tool to create a base image, and inject the materials you want in to that.

On the user experience: I've been actively working to make the experience identical on the various platforms Raspberry Pi Imager operates on - because users, entirely fairly, expect that the tool will do the same thing regardless of their choice of host OS. Introducing new features that may only appear on Linux hosts (for likely a long time) creates two tiers of user - and that is absolutely not something I'm prepared to do. Penalising a user for their choice of host OS is not an acceptable choice in this project.

While I understand that this rejection may be frustrating, I would encourage you to consider my points. Equally, if you are able to demonstrate an implementation that works on macOS, Windows and Linux, I am prepared to review a PR.

closed 2024-08-30 by tdewey-rpi

roblatour, 2024-08-30:
Thank you for your time and consideration.